raspberrypi/rpi-imager #310: Advanced Options: Enable SSH?

saltedlolly, 2021-11-20:
Would it be possible to include a tickbox to enable SSH in the advanced settings? This could be as simple as having the imager place the 'ssh' file in the root folder after the image has been flashed successfully. It would remove the extra steps required to add it when doing a headless setup. Bonus if it could optionally also install a avahi-daemon automatically as well.

maxnet, 2021-11-20:
We already have a box for SSH.
And think you tried to use it before, judging from your previous issue: :-)

>When trying to use advanced mode to *enable SSH*, set hostname, password etc

==

>Bonus if it could optionally also install a avahi-daemon automatically as well.

Currently not so optional.
Current code assumes you want avahi if you specify a hostname (as our advanced settings dialog box shows ".local" at the end of the hostname)

(There do is a slight issue with cloud-init package installation on Ubuntu, that could potentially cause avahi and other extra package installation to fail at some point in the future.
It runs "apt update", but that fails, because it is executed too early, at a time the Pi does not have the correct date/time because NTP time synchronization is not complete yet, failing their checks.
It does subsequentally succeed in installing avahi, using the existing repository information that was bundled in the Ubuntu image.
But if that gets out of date, and they remove older package files from their repository, it could cause problems in the future.
Filled a bug: https://bugs.launchpad.net/bugs/1951639 )

maxnet, 2021-11-20:
BTW if you are using wifi, you need Ubuntu 21.10.
The older Ubuntu 21.04 LTS has a bug in which wifi will not work until after first reboot.

closed 2023-10-19 by cillian64