raspberrypi/rpi-imager #311: Advanced Options: Enable VNC SERVER ???

karadevnet, 2021-11-21:
hi, i install BULLSEYE and it works fine at PI3B and little overclock safety also is prove work of my system. BUT for next update ot BULLSEYE i recommend strongly to make a option ENABLE VNC server in option menu in PI imager v1.6.xxxx . because is so hard always to change HDMI cable from my monitor of other system to pi , make first install seting and it is just for to enable VNC SERVER in PI settings. we are already in 21 century, what is problem to make it option VNC enable like a SSH enable option in install menager PI imager ???

and also if you can make all text to be purr black color because greay on white is not read well !!!! can you do this cool stuff like you do cool RPI 2040 PICO and all RPI4 8GB ??? can you ???? thanks very mach about understanding of my ask. best regards to all programmers from i programmer :)

here some picture for what i want to be more easy and good for all users
[image]

lurch, 2021-11-22:
As a "workaround" I guess you could use the "enable SSH" option, then [ssh to your Pi](https://www.raspberrypi.com/documentation/computers/remote-access.html#secure-shell-from-windows-10) and then [use `raspi-config` to enable VNC](https://www.raspberrypi.com/documentation/computers/remote-access.html#enabling-vnc-server-at-the-command-line) ?

maxnet, 2021-11-22:
Downside of offering VNC is that not everyone is allowed to use it, and explaining the restrictions would take up a lot of GUI footprint.

(Commercial users and anyone using it on more than 10 Pi (e.g. schools) are supposed to purchase a paid license.
(See RealVNC's small print: https://static.realvnc.com/media/documents/LICENSE-4.0-raspi_en.pdf )

SSH has no such restrictions.

ghollingworth, 2021-11-22:
Are there any other options for VNC server which don't have this limitation (tightvncserver for example?)  

Although I don't believe this works in exactly the same way (it creates a new separate DISPLAY rather than 'controlling' the main display.

Gordon

XECDesign, 2021-11-22:
x11vnc uses the real display

maxnet, 2021-11-22:
>Are there any other options for VNC server which don't have this limitation (tightvncserver for example?)

x11server
But recall it needs extra configuration to set password (instead of using system's through PAM).

binsearchnet, 2021-11-29:
Tried x11vnc, but it is absolutely unusable under the default dtoverlay=vc4-kms-v3d
Not only does graphics lag behind, it is in fact so slow that if you open Chrome and press a key on the keyboard to enter an URL, you get the key repeated, because it so busy with the graphic side of things, that processing the KEY_UP event does not make it in time.
(Does work correctly without KMS)

maxnet, 2021-11-29:
Tried x11vnc, but it is absolutely unusable under the default dtoverlay=vc4-kms-v3d
Not only does graphics lag behind, it is in fact so slow that if you open Chrome and press a key on the keyboard to enter an URL, you get the key repeated, because it so busy with the graphic side of things, that processing the KEY_UP event does not make it in time.
(Does work correctly without KMS)

lawrence-jeff, 2022-04-01:
> Downside of offering VNC is that not everyone is allowed to use it, and explaining the restrictions would take up a lot of GUI footprint.
> 
> (Commercial users and anyone using it on more than 10 Pi (e.g. schools) are supposed to purchase a paid license. See RealVNC's small print: https://static.realvnc.com/media/documents/LICENSE-4.0-raspi_en.pdf )
> 
> SSH is free for everyone.

I'm not sure I understand this concern, these restrictions aren't mentioned when you enable VNC via raspi-config so is there a concern enabling it in the imager UI. This would be a big improvement, having SSH is nice but the extra step of going in and running raspi-config just to enable VNC seems unnecessary

lurch, 2022-04-03:
> having SSH is nice but the extra step of going in and running raspi-config just to enable VNC seems unnecessary

OTOH, with SSH enabled as kind of a "minimum connectivity feature", you _can_ then still do what you need to. If we added an option to also enable VNC, that might become a slippery slope with other people asking for RDP to be enabled, or TeamViewer to be enabled, etc. etc. ? :man_shrugging:

lawrence-jeff, 2022-04-03:
> OTOH, with SSH enabled as kind of a "minimum connectivity feature", you _can_ then still do what you need to. If we added an option to also enable VNC, that might become a slippery slope with other [...]

A bit of apples and oranges there- VNC is one of the integrated options, which is completely different than random solutions like RDP or Team Viewer that aren't even in the OS. It doesn't seem a bad goal to expose more of the Raspi-config options at the earlier stage. There are scenarios like classes/makerspaces where the ability to build a bunch of images that are pretty close to being ready to go is desirable. Yes I know you can instruct students to do the SSH steps but sometimes these are kids or not very computer literate folks and your just trying to show them the basics

tdewey-rpi, 2024-02-06:
Rejecting feature request.

Unfortunately this tickbox requires a range of infrastructure not available today - not least, we have to confirm the target OS can actually provide a VNC server - something that would likely require a change to the JSON schema to accommodate.

While rpi-imager isn't the right place for this, alternative tools might offer a better solution. In particular, packaging a VNC server with the OS you flash via 'Use Custom' might offer a mass-deployable solution in the immediate term. Additionally, I may look at creating alternative tooling to support this case, but I can't offer a guarantee at this time.

closed 2024-02-06 by tdewey-rpi