raspberrypi/rpi-imager #312: Server Connection Issue?

ahdeck, 2021-11-25:
Am I the only one having this issue? It's been going on for a couple of days. I'm not having any internet issues on my end as far as I know, and my download speed it currently 175mbps. Maybe it's just a server problem for them, to be resolved after the weekend?

[image]

maxnet, 2021-11-25:
Seems it is trying to connect over IPv6 in your case.

https://test-ipv6.com/ shows your IPv6 connectivity is good?

ahdeck, 2021-11-26:
Thanks for your response! And it looks like you are onto something. When I run the test, everything passes except for the last category:
"Your DNS server (possibly run by your ISP) appears to have no access to the IPv6 Internet, or is not configured to use it. This may in the future restrict your ability to reach IPv6-only sites."

So, I guess the next question is...how do I get the imager to use a compatible protocol? I am just a noob with a pi.

maxnet, 2021-11-26:
Try if it work better if you disable IPv6 globally: https://networking.grok.lsu.edu/article.aspx?articleid=17573

ahdeck, 2021-11-30:
Hope it was a good holiday.

That was easy to do. Unfortunately, no dice.

[image]

I'm going to be upgrading my internet equipment this week. Let me try again after that. That still seems strange though, it's not like I'm on dial-up or satellite from a remote part of Alaska. So, if you think of anything else in the meantime, please let me know.

Thank you for your help as always.

maxnet, 2021-11-30:
>Hope it was a good holiday.

Am afraid not so in Europe.
(Most get plenty of vacation days to make up for our lack of national holidays though)

>That was easy to do. Unfortunately, no dice.

Are you sure you only unticked IPv6 (and not accidently IPv4 as well)?

ahdeck, 2021-12-01:
Hadn't disabled IPv4, but with a new wireless adapter (USB AC600 to PCE AC1900) the problem seems to be solved. ¯\_(ツ)_/¯

Thanks for all your help!

tdewey-rpi, 2024-02-06:
Closing as won't fix - problem appears to have been related to a HW configuration.

closed 2024-02-06 by tdewey-rpi