raspberrypi/rpi-imager #313: How to translate?

miharix, 2021-11-27:
Hi,

can some one help me? I want to translate rpi-imager to sl-SI
 but I seem to make some stupid beginner mistake because I'm unable to compile.

OS: ubuntu 21.10 - fresh install
git clone on date 27.11.21
my steps after steps in README.md
```
cd i18n 
cp rpi-imager_en.ts rpi-imager_si.ts
linguist rpi-imager_en.ts
```
"Edit" -> "Translation File Settings..." -> Language -> "Slovenian (slovenščina)" ->Country/Region -> "Slovenia (slovenščina)" -> OK
Go to context "main"
Translate in Strings "CHOSE OS" -> "mark item as done"
Save -> Exit
`nano translations.qrc`
Add line 
`<file>rpi-imager_si.qm</file>`
```
cd ..
debuild -uc -us
```

```
[  8%] Generating rpi-imager_sk.qm
[ 11%] Generating rpi-imager_zh.qm
[ 14%] Generating rpi-imager_it.qm
/usr/lib/qt5/bin/lrelease /home/miharix/gitc/rpi-imager/i18n/rpi-imager_sk.ts -qm /home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_sk.qm
/usr/lib/qt5/bin/lrelease /home/miharix/gitc/rpi-imager/i18n/rpi-imager_zh.ts -qm /home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_zh.qm
/usr/lib/qt5/bin/lrelease /home/miharix/gitc/rpi-imager/i18n/rpi-imager_it.ts -qm /home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_it.qm
Updating '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_de.qm'...
Updating '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_sk.qm'...
Updating '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_zh.qm'...
Updating '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu/rpi-imager_it.qm'...
    Generated 125 translation(s) (125 finished and 0 unfinished)
    Ignored 9 untranslated source text(s)
    Generated 125 translation(s) (125 finished and 0 unfinished)
    Ignored 9 untranslated source text(s)
    Generated 127 translation(s) (127 finished and 0 unfinished)
    Ignored 7 untranslated source text(s)
    Generated 125 translation(s) (125 finished and 0 unfinished)
    Ignored 9 untranslated source text(s)
make[3]: Leaving directory '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:98: CMakeFiles/rpi-imager.dir/all] Error 2
make[2]: Leaving directory '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu'
make[1]: *** [Makefile:152: all] Error 2
make[1]: Leaving directory '/home/miharix/gitc/rpi-imager/obj-x86_64-linux-gnu'
dh_auto_build: error: cd obj-x86_64-linux-gnu && make -j4 VERBOSE=1 returned exit code 2
make: *** [debian/rules:7: build] Error 25
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

If I don't edit translations.qrc, it compiles, but translation is not visible after
```
cd ..
sudo apt remove rpi-imager
sudo apt install ./rpi-imager*.deb
```

maxnet, 2021-11-28:
In CMakeLists.txt there are two lines of interest:

```
#qt5_create_translation(QM_FILES ${CMAKE_SOURCE_DIR} i18n/rpi-imager_en.ts i18n/rpi-imager_nl.ts i18n/rpi-imager_zh.ts i18n/rpi-imager_tr.ts i18n/rpi-imager_fr.ts i18n/rpi-imager_de.ts i18n/rpi-imager_sk.ts i18n/rpi-imager_it.ts i18n/rpi-imager_ca.ts)
qt5_add_translation(QM_FILES i18n/rpi-imager_en.ts i18n/rpi-imager_nl.ts i18n/rpi-imager_zh.ts i18n/rpi-imager_tr.ts i18n/rpi-imager_fr.ts i18n/rpi-imager_de.ts i18n/rpi-imager_sk.ts i18n/rpi-imager_it.ts i18n/rpi-imager_ca.ts)
```

- Add imager_sl.ts to both.
- Uncomment (remove the #) from the qt5_create_translation line, build, add the # again.

Should generate a .ts you can edit with the Qt languist tools.

(Make sure you comment the qt5_create_translation line afterwards, as it has the tendency to delete the .ts file again on a "make clean" otherwise).

miharix, 2021-11-28:
Worked :smiley: Thanks!

closed 2021-11-28 by miharix