raspberrypi/rpi-imager #315: OS List - multi language support

miharix, 2021-11-28:
I would purpose to implement multiple language support for
[https://downloads.raspberrypi.org/os_list_imagingutility_v2.json](https://downloads.raspberrypi.org/os_list_imagingutility_v2.json)

rpi-imager in non English language looks incompletely translated because of this list.

maxnet, 2021-11-28:
Problem is the entries in the OS list are maintained by a bunch of different people.
We link to other files from the main one.

(If Linux distributions want to display their own distribution in another language, they can already put alternative "os_list_LANGUAGECODE" sections in their json BTW)

lurch, 2021-11-28:
> (If Linux distributions want to display their own distribution in another language, they can already put alternative "os_list_LANGUAGECODE" sections in their json BTW)

Interesting - perhaps that should be documented in the `README.md` ?

tdewey-rpi, 2024-02-06:
Suggestion accepted.

I'll take a look at some documentation around this later in the week, as it also annoyed me when forcing the tool in to French.