raspberrypi/rpi-imager #316: Adding SSH public keys to "set authorized_keys for pi" in advanced settings doesn't seem to work

raspberrycoulis, 2021-11-30:
I've been using SSH keys to authenticate with my Raspberry Pi's for a long time now, and until it was supported on the Raspberry Pi Imager I would usually copy a bash file to my SD card that would disable password authentication, turn on public key authentication and then copy my SSH public keys to the `authorized_keys` file before restarting the SSH service within a few seconds.

However, I have now seen that you can paste the SSH public keys to the setting called "set authorized_keys for pi" under the advanced settings menu, but despite pasting the same contents to that box, it does not work when I boot my newly flashed Raspberry Pi.

Does the field in the advanced settings allow for multiple SSH public keys to be added (it is suggested given it's "set authorized_key**s**")? Based on my experience, it does not seem to be working as expected as I cannot SSH into my Raspberry Pi whenever I use this setting, meaning I can only revert back to my previous method which is to manually add my bash script to my SD card, then run it via SSH via the default password, to set it up how I like.

maxnet, 2021-11-30:
Only supports single line in current version.

geerlingguy, 2021-12-20:
I had asked for support for _multiple SSH keys_ in #326, but that issue was marked as a dupe of this one, so I'm subscribing to this issue to ask for it as well. I think the title of this issue is a little off, though—if I paste in a file full of pubkeys, the entire content is dumped into the authorized_keys file with a space between each newline. So the first key pasted does work, but others don't, until you manually go in and put all the separate keys on separate newlines again.

So my feature request would be for the public key field in Imager to support newline characters.

closed 2021-12-20 by maxnet

pascal456, 2022-11-20:
This has been closed without comment. I have a question on the changes in the linked commit. Is it correct that multiple keys have to be added in the single-line input field? And that they have to be separated by `\n`?

maxnet, 2022-11-20:
Don't think there is documentation for that. Had assumed the audience that is using multiple keys, doesn't read docs.

\n = new line

It is assuming you already have a text file with all your public keys, one per line, and are just copying and pasting that to Imager in that single field, in one go.

commit 47d331b6 (2021-12-20): Allow multi-line authorized_keys

Closes #316