raspberrypi/rpi-imager #318: Forced WiFi Password

reukiodo, 2021-12-02:
### Steps to reproduce:
1. set up open WiFi AP
2. run rpi-imager
3. CTRL+SHIFT+X
4. check 'Configure wifi' box
5. set SSID to open WiFi AP SSID
6. click SAVE button

### Expected:
Saves settings and goes back to main screen.

### Actual:
Refuses to accept custom settings, highlighting 'Password: ' and its field red as if they are required.

reukiodo, 2021-12-02:
Related: #188

reukiodo, 2021-12-02:
This is covered with the `key_mgmt=NONE` in wpa_supplicant.conf, as shown from https://howchoo.com/g/ndy1zte2yjn/how-to-set-up-wifi-on-your-raspberry-pi-without-ethernet

maxnet, 2021-12-02:
Not sure if open wifi is something we should be promoting.
Keep in mind the wifi password provides mutual authentication.
No way to know if the access point you are connecting to is really your own access point with open wifi.

ghollingworth, 2021-12-02:
Agree, this should not be an option in rpi-imager.  If you want to enable this you should do so post-boot.

closed 2021-12-02 by ghollingworth

reukiodo, 2022-04-15:
Are you the WiFi police now? Just because you think it isn't a good idea doesn't mean that others don't depend on it. There are open WiFi networks all around the world and not even allowing advanced config to connect to them is limiting a lot of uses.