raspberrypi/rpi-imager #322: RPi imager does not resume downloads ?

shodanx2, 2021-12-07:
Hi,
I'm on very slow internet, I tried to write rpios twice but it ends up failing.

I suspect this is either because of network interruption or my computer simple going to sleep.

Where does RPi Image store partial download ? 

Is it supposed to be able to resume these download in case of error during download ?

maxnet, 2021-12-08:
- It tries to resume downloads on interruptions that occur when a computer is running.
- It is unable to resume if it does go down.
- On Windows it prevents computers going to sleep. (To the extend that is possible. Naturally, if you did not plug your laptop in, and the battery runs out, there is little we can do to keep the computer running...)

shodanx2, 2021-12-09:
Sorry, I missed the error message.
And now that I have a finished the download, I cannot reproduce.

I have just selected another OS (turns out I can't use bullseye for my application, due to missing python2)
and one thing that is confusing me is, as soon as you click write it shows "writing" 

It did show "downloading"  for a few seconds but that wasn't anywhere near long enough to have downloaded that 1GB file !

So, the UI not making the distinction between writing or downloading, is making it hard for me to figure out what is going on.
In particular, I couldn't tell earlier if the actual writing process was slow, or if it was the downloading, if it was the downloading, I couldn't tell if it successfully resumed the already downloaded portion or if it started over ?

[image]

Where does RPi Imager download the images ?

I found this folder but can't tell where it puts the file, whether they're growing or if they're getting reset to zero on retries

[image]

maxnet, 2021-12-13:
>Where does RPi Imager download the images ?

If you have plenty of disk space there should be a lastdownload.cache file a couple folders up from where you are looking.
Exact location can differ slightly depending on operating system and version.

If you are low on space it does not cache files.
It is technically able to stream the download straight to your SD card, uncompressing on the fly, without having to save it to disk first.

It also slows down the download if your storage write speed is slow.
So no, there is no easy way to tell which is causing slow downs.

Just download the image file manually from: https://downloads.raspberrypi.org/raspios_armhf/images/
And feed the file to Imager with the "use custom" option to see. (No need to uncompress the .zip first).

tdewey-rpi, 2023-11-17:
Closing as stale.

closed 2023-11-17 by tdewey-rpi