raspberrypi/rpi-imager #323: Verifying write failed...

Aviators01, 2021-12-08:
When trying to flash a OpenWrt .img.gz image file to a 128gb SDxc class 10 microSD card, it gives the characterisitic "Verifying write failed. Contents of SD card is different then than what was written to it." Balena Etcher does it without problems.

maxnet, 2021-12-08:
Is verification enabled in Etcher?

Aviators01, 2021-12-09:
Yes, Etcher does verification. I don't think there is an option to enable/disable it.

maxnet, 2021-12-13:
>Yes, Etcher does verification. I don't think there is an option to enable/disable it.

There used to be, but seems they removed it in newer versions.

BTW did you also try flashing larger images than OpenWrt to the same card with Etcher, or ran H2testw to double check there is nothing wrong with storage?

Because I am not sure whether or not the way Etcher performs verification guarantees that the data is actually read from the card, instead of memory disk cache.
OpenWrt images are typically only a small number of MB, right?!

Also what operating are you using?

Aviators01, 2021-12-24:
I have flashed raspberry pi os to it before. I'm running the tester currently.

I'm not sure. When I try to flash OpenWrt with the imager, openwrt doesn't run besides not verifying. Etcher manages to flash it but I'm also having trouble with getting OpenWrt to run properly, so I'm not sure where exactly the problem is. OpenWrt uses only 107MB.

>Also are you using Windows, or are you running Imager under another operating system?

I'm using Windows.

I bought another microsd so I'll attempt flashing openwrt to the new one next week.

maxnet, 2021-12-24:
>When I try to flash OpenWrt with the imager, openwrt doesn't run besides not verifying.

That's by design.
We zero out the partition table when starting to write, to prevent other applications from accessing the files on card while we are still writing and verifying.

And if it does not verify successfully, we will not finish up and write the real partition table, preventing you using the card that did not verify.

Aviators01, 2021-12-30:
ok. So I redownloaded the OpenWrt image file and flashed it and there wasn't any problems this time. I'm not sure why as I checked the sha256 sum both times and they were identical. I had previously flash raspberry pi os to it so I knew it worked. Anyways, thanks for your help.

tdewey-rpi, 2024-02-06:
Closing as stale - it also sounds like the OP managed to get to success.

closed 2024-02-06 by tdewey-rpi