raspberrypi/rpi-imager #324: SSH does not work with ubuntu

aik, 2021-12-14:
I programmed the sd card with ubuntu 2104, crtl-shift-x, ticked ssh and the key, flashed, booted, ssh is running but not accepting the key.
I put the card back to the PC and discovered that "writeable" partition does not have the "pi" use at all and no authorized_key contains anything, I added those, rebooted, works.

So is the "pi" user and ssh setup only supposed to work with debian? The docs don't seem mention it. Thanks,

lurch, 2021-12-14:
Duplicate of #213 , #286 etc. ?

maxnet, 2021-12-14:
If you compile Imager from git, and "use custom" to select a local Ubuntu server image, you do can customize that now.
(Not the desktop edition)

aik, 2021-12-15:
> Duplicate of #213 , #286 etc. ?

yeah, sorry for the noise.

> If you compile Imager from git, and "use custom" to select a local Ubuntu server image, you do can customize that now. (Not the desktop edition)

ctrl-shift-x works when build from git and the image is local and not downloaded? Or you meant I can change the image myself before telling rpi-imager to write it to the card? If it is that, fixing the image afterwards seems easier (my PC runs fedora anyway).

maxnet, 2021-12-15:
>ctrl-shift-x works when build from git and the image is local and not downloaded?

1) If you download Ubuntu **server** from the Internet manually. E.g. by going to: http://cdimage.ubuntu.com/releases/impish/release/ubuntu-21.10-preinstalled-server-arm64+raspi.img.xz

2) And open that in the latest Imager, you can use advanced settings. It should also show a cog icon on the bottom right that gives you the settings:

[image]

Will not work with images from the repository right now, as the repository files has not been updated yet to flag correctly which operating system supports customization and which does not.
(With local files it attempts to auto-detect).

Also only supports Ubuntu Server, NOT Ubuntu Desktop.

guysoft, 2022-01-11:
Hey, can confirm. 
Thanks for pointing out why it does not work. I kept downloading my custom distros and being surprised that it works but then that it does not work on the official images.

cillian64, 2023-10-19:
Closing as a dupe of #213

closed 2023-10-19 by cillian64