raspberrypi/rpi-imager #325: Avoid using the discrete GPU on Macs

3rror, 2021-12-16:
Raspberry Pi Imager forces the use of the discrete GPU on the MacBook Pro 16" with Intel processor and AMD GPU, and probably on all Macs with two graphic units.

It's probably because it is missing the `NSSupportsAutomaticGraphicsSwitching` key in `Info.plist`.

```xml
<dict>
	<key>NSSupportsAutomaticGraphicsSwitching</key>
	<true/>
	[...]
```

Setting it to `true` maybe fix the issues. More info can be found here: https://developer.apple.com/library/archive/qa/qa1734/_index.html

maxnet, 2021-12-16:
>Setting it to true maybe fix the issues.

I am not that familiar with Apple gear, and do not have a Macbook (only a simple Mac mini for development purposes)
So can you spell out what those issues are exactly?

Why is it a bad thing that our application -that indeed uses OpenGL graphics acceleration- uses the discrete GPU?
Is it just a power consumption thing, that would make it better if it could use the Intel GPU, or are there more serious issues?

3rror, 2021-12-16:
> Why is it a bad thing that our application -that indeed uses OpenGL graphics acceleration- uses the discrete GPU? Is it just a power consumption thing, that would make it better if it could use the Intel GPU, or are there more serious issues?

It's just to save battery life, and also reduce the heat produced resulting in less fan noise when combined with other intensive tasks.
Nothing _really serious_ in practice 😃. I noticed it because it happened to leave the tool open while writing multiple images.

I  have to check if I can build the Imager locally and see if the fix works. If I remember correctly, I have already applied it in the past to some Qt apps.

maxnet, 2022-01-23:
Can you try if this build solves the issue on your hardware?
(while modifying the plist is easy enough. it is untested as I don't have dual GPU gear)

[imager-plist.dmg](https://www.maxnet.eu/tmp/imager-plist.dmg)

3rror, 2022-01-23:
Sorry, I haven't used the Imager in the last few weeks so I forgot about the issue.
Yeah, I just tested it and it works. It doesn't force the discrete GPU anymore 😄.

commit 51a79c72 (2022-01-23): Mac: set NSSupportsAutomaticGraphicsSwitching in plist

Closes #325
closed 2022-01-23 by maxnet