raspberrypi/rpi-imager #326: Support for multiple SSH pubkeys

geerlingguy, 2021-12-20:
One thing I would like to do sometimes is to add more than one SSH public key to the Pi via imager at a time.

Currently if I go into advanced settings and paste two in (with a newline separating them), they get pasted with a space between the newlines. Then I have to boot the Pi, log into it using the first public key, then edit `~/.ssh/authorized_keys`, and separate the two lines by hand.

It would be a convenience, but understandably not a priority, since I'm probably one of few people who manage multiple SSH keys _and_ do so with a bunch of Pi infrastructure :)

maxnet, 2021-12-20:
Closing this as duplicate of #316
(And recall there is also a somewhat related third issue somewhere with suggestion to allow selection of a file, instead of copy & pasting)

closed 2021-12-20 by maxnet

geerlingguy, 2021-12-20:
@maxnet ah, based on the title of that other issue I thought it was just someone having trouble with that field. I do see it works, but only for the first key pasted.