raspberrypi/rpi-imager #328: Imager 1.6.2 for Mac: No OS options only erase or custom install

exidyboy, 2021-12-29:
Imager 1.6.2 for Mac: No OS options only erase or custom instal
[image]
l

maxnet, 2021-12-29:
Any proxy server configured in Mac OS, or anything else out-of-the-ordinary network wise?

exidyboy, 2021-12-29:
Yes I am behind a proxy. I didn't realise the Imager needed Internet access to populate the list of options. Perhaps it could display an error if there is a network problem preventing it from accessing the resources it needs?

maxnet, 2021-12-29:
It do is supposed to show an error once it fails.
Not sure if that always works with proxy servers though, e.g. if they just take forever to respond instead of failing.

exidyboy, 2021-12-31:
I timed it and if you are behind the proxy (where I work) you get an error  that says "Error downloading OS list from the Internet" after 12 minutes. Perhaps it could fail more gracefully.
[image]

closed 2021-12-31 by exidyboy

maxnet, 2021-12-31:
Yes, might be 2 Qt bugs in here.

1) Qt does have does have code to automatically fetch and use the proxy settings of your operating systems.
Not sure why it does not work.
Could be that the more advanced proxy types (e.g. the proxies that require authentication) are not implemented.

2) We do set a network timeout of 5 seconds.
So normally you do should get that error really fast.
But it wouldn't surprised me if the timeout code is not working correctly in the more advanced corner cases. Like when proxies are used.