raspberrypi/rpi-imager #329: OS List: Addition of nymea and/or BerryLan images

mzanetti, 2022-01-07:
Would it be possible to add the nymea image to the OS list? nymea (https://nymea.io) is an open source home automation system and IoT platform. It offers C++ development libraries and services tailored for use within an IoT context, reaching from headless WiFi setup for Raspberry Pis to a full ZigBee stack supporting the phoscon ZigBee Raspberry Pi dongle. Putting all those pieces together, the nymea community image offers a fully featured and easy to use home automation system (100% open source on github: https://github.com/nymea).

BerryLan (https://berrylan.org) is a breakout of the WiFi setup feature as a standalone image as it has been received rather well by users for all sorts of use cases. It is based on the upstream Raspberry Pi OS images, just adding the minimal changes required to allow provisioning WiFi credentials to a Raspberry in a headless manner by implementing a Bluetooth service and the according app for it (found in the mobile device app store of your choice - also FOSS: https://github.com/nymea/berrylan/).

For more information on the images, please see nymeas fork of the pi-gen repository: https://github.com/nymea/pi-gen (berrylan and nymea branches) - There are pull requests showing the diff towards the upstream Raspberry Pi images for each of them.

lurch, 2022-01-08:
I believe each of those projects would need to put together (and host) a suitable JSON file and then request inclusion into RPi Imager themselves? See e.g. #219

ghollingworth, 2022-01-08:
I think the WiFi setup isn't much use, considering rpi-imager already has this functionality...

mzanetti, 2022-01-08:
> I think the WiFi setup isn't much use, considering rpi-imager already has this functionality...

I agree that if you're just thinking the use case of downloading the image and setting up something for yourself this isn't of much use:

* Having demo Raspberry Pis with some certain prepared setup and showing it off at a talk (or other demo) and easily allow connecting it to the WiFi.
* Preparing a Raspberry Pi for someone else and handing it over to someone who doesn't have the skills to edit network config files via SSH.
* Building actual products for people who don't know that a Raspberry Pi is inside (Just one example pishock (NSFW) uses berrylan)

Anyhow, not trying to argue, just to clarify what the intentions of BerryLan are as it's often misinterpreted.

> I believe each of those projects would need to put together (and host) a suitable JSON file and then request inclusion into RPi Imager themselves? See e.g. #219

Ah, thank you! Will try to do that then.

mzanetti, 2022-01-09:
```{
  "name": "nymea",
  "description": "Open-Source Smart Home/IoT platform with focus on ease of use and privacy.",
  "icon": "https://raw.githubusercontent.com/nymea/nymea/master/icons/nymea-logo-48x48.png",
  "subitems_url": "https://nymea.io/rpi-image-repo.json"
}```

Ok, I've created the json file on the nymea server. Currently only for the full nymea image.

ghollingworth, 2022-02-02:
Can you email me at gordon@ this organisation dot com

ghollingworth, 2022-02-02:
Added

closed 2022-02-02 by ghollingworth