raspberrypi/rpi-imager #330: [OS Request] OctoPi-Klipper

guysoft, 2022-01-09:
Hey,
I am the maintainer of OctoPi-Klipper and OctoPi.
OctoPi addition went pretty well, I wish to add OctoPi-Klipper too.
There is an issue with the manual install documented on Klippers issue tracker here: https://github.com/Klipper3d/klipper/pull/4635 
Publishing the images makes it accessiable as mainline OctoPi.

Following instructions as here: https://github.com/raspberrypi/rpi-imager/issues/162

Note that it should go in the ``Other spesific purpose OS`` section.
```json
{
    "os_list": [
        {
            "name": "OctoPi-Klipper",
            "description": "Includes the OctoPrint host software for 3d printers and Klipper firmware service",
            "icon": "https://raw.githubusercontent.com/guysoft/OctoPi/devel/media/rpi-imager-OctoPi-klipper.png",
            "subitems_url": "http://unofficialpi.org/rpi-imager/rpi-imager-octopi-klipper.json"
        }
    ]
}
```

And the json file is in place:
http://unofficialpi.org/rpi-imager/rpi-imager-octopi-klipper.json

The json file is built automatically using github actions now :)

guysoft, 2022-02-02:
This is a sub-category of OctoPi list now.
[image]

ghollingworth, 2022-02-02:
I don't think you can do this.  Version 1.6 of RPi-Imager will crash when you select this menu...

guysoft, 2022-02-02:
@ghollingworth It seems to work for me on 1.6.2, and the one that introduced it was @foosel so I assume it works for her too.
In what scenario is it expected to crash?

ghollingworth, 2022-02-02:
https://github.com/raspberrypi/rpi-imager/commit/c17795c48eeed597b9e62e11ddc74a5557b8384b

Did you test it on all platforms?

foosel, 2022-02-02:
I tested on the platform I have access to (Windows) and assumed since it worked there it was fine and supported.

Can roll it back for now to not cause crashes, if that is intended happening on other platforms, but for cases like this it would be good if there was some kind of JSON schema or some other form of dependable spec for the supported JSON structure, since apparently it is not self documenting and recursive (as in, looking at the master document does not hint at what works in sub trees).

maxnet, 2022-02-02:
It is fine since "subitems_url" is used.
Problem was with nested inline "subitems" nodes.

foosel, 2022-02-02:
Thank you for clarifying @maxnet!

ghollingworth, 2022-02-02:
Ah ok,

So we'll remove the upper level Klipper entry

foosel, 2022-02-02:
Please don't confuse Klipper with Octo-Klipper 😅 Octo-Klipper did never made it into your main json file last I checked, which is why I put it into the OctoPi one to help out Guy. Octo-Klipper is OctoPrint with Klipper, but I think there might also be a Klipper-only image, possibly with other frontend alternatives on board.

ghollingworth, 2022-02-02:
Well, everything is now in rpi-image if you check (i.e. as of the last hour or so...)

Gordon

XECDesign, 2022-02-02:
All of these are not in the v2 json file, but if they were meant to be, I'll add them now.

XECDesign, 2022-02-02:
Give it a go now. I've added the OctoPi-Klipper entry as well, so it can be removed from OctoPi's json.

I can understand the good intentions, but please don't use your own json files to circumvent the right process.

ghollingworth, 2022-02-02:
Good point, the v2 file cannot have the sub-menu.  Those will have to wait for a couple of days until we release a new rpi-imager update

foosel, 2022-02-02:
> Give it a go now. I've added the OctoPi-Klipper entry as well, so it can be removed from OctoPi's json.

Can confirm. And the submenu-url from OctoPi's json should be gone too, see the above commit.

> I can understand the good intentions, but please don't use your own json files to circumvent the right process.

Noted. It was merely an attempt at helping out, no harm intended :)

guysoft, 2022-02-02:
@ghollingworth Suggestion: If you guys have some kind of documentation, or even open the wiki here so we could document what should work, it would be **spec**-tacular :)

@XECDesign We have no intention to go against due process. After the comment that this can actually break something @foosel removed the klipper entry. And after @maxnet 's comment that it should be fine it was added back again. We don't enjoy doing multiple work as do you.

foosel, 2022-02-02:
Quick correction to that, I didn't remove things in between these two time points as I was still assessing the situation and the "all good" from @maxnet came before I did the deed. But now the OctoPi-Klipper subentry is gone, the nightlies however are still in there (as a subitem url, not as subitems).

guysoft, 2022-03-09:
Hey, so sorry to bug but we need to update.

Due to OctoPi-Klipper having both a name and a logo looking like OctoPi and the list getting randomly re-arranged. We have decided to change the name and logo because some users were getting confused. Related: https://github.com/guysoft/OctoKlipperPi/issues/4

new json:

```json
{
    "os_list": [
        {
            "name": "OctoKlipperPi",
            "description": "Includes the OctoPrint host software for 3d printers and Klipper 3D printer firmware service",
            "website": "https://github.com/guysoft/OctoKlipperPi",
            "icon": "https://github.com/guysoft/OctoKlipperPi/raw/main/media/rpi-imager-OctoKlipperPi.png",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-octoklipperpi.json"
        }
    ]
}
```

XECDesign, 2022-03-10:
The new snippet should be in now.

guysoft, 2022-03-10:
Confirmed, closing :)

closed 2022-03-10 by guysoft