raspberrypi/rpi-imager #333: [OS Request] Gladys Assistant, a privacy-first, open-source home assistant

Pierre-Gilles, 2022-01-11:
Hi,

I'm the maintainer of [Gladys Assistant](https://gladysassistant.com/), an open-source home automation software that runs on the Raspberry Pi primarily.

We distribute a Raspberry Pi OS image with Gladys Assistant pre-installed, and would love to see it in the amazing Raspberry Pi Imager ! 

```
{
    "os_list": [
        {
            "name": "Gladys Assistant",
            "description": "A privacy-first, open-source home assistant that runs on the Raspberry Pi.",
            "icon": "https://gladysassistant.com/img/external/raspberry-pi-imager/icon-raspberry-pi-imager-40x40.png",
            "subitems_url": "https://gladysassistant.com/img/external/raspberry-pi-imager/raspberry-pi-imager-gladys-assistant.json"
        }
    ]
}
``` 

Don't hesitate if you see a mistake or if I need to change something. 

Thanks again for building this tool, it's a really good one 👍  I personally love the "set Wi-Fi in advance" feature, without any file edit required !

ghollingworth, 2022-02-02:
Can you email me at gordon@ this organisation dot com

Pierre-Gilles, 2022-02-02:
Thanks for the merge. I just sent you an email !

Pierre-Gilles, 2022-02-07:
@ghollingworth I just tried the download & clone feature, and I get this error with Gladys Assistant image : 

[image]

It works with other images... Any idea what's going on ?

I suspect it because the download URL we submit is not a direct download link, there is a redirect (we use a generic URL that redirects to the latest version). 

Do rpi-imager follow redirects ?

Pierre-Gilles, 2022-02-07:
In CLI, the download seems to work: 

```
pierregilles@PierreGllessMBP ~ % curl -L --max-redirs 10 https://gladysassistant.com/download/latest --output test.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    67  100    67    0     0    112      0 --:--:-- --:--:-- --:--:--   113
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
  0  524M    0 3520k    0     0  2414k      0  0:03:42  0:00:01  0:03:41 2414k^C
```

XECDesign, 2022-02-07:
```
Hash of uncompressed image: "8ffc768a9659a497b1bbc0cce83d7b448a8d327c8a0d5dd0b9471374ea52befd"
Mismatch with expected hash: "f7848bfb518f00c6af225bbd2d4a05121961f45fa7a3e0f42be81c8964e260a4"
```

That 'uncompressed hash' matches the hash of the zip file. I thought maybe the json file had the values flipped, but no, it looks okay.

Maybe rpi-imager sees 'latest' in the URL without any extension and then treats it as a raw uncompressed image rather than a zip file?

Pierre-Gilles, 2022-02-07:
> Maybe rpi-imager sees 'latest' in the URL without any extension and then treats it as a raw uncompressed image rather than a zip file?

Oh ! Should I change something on my side? (Like "latest.zip"), or is it something that should be fixed in rpi-imager ?

Pierre-Gilles, 2022-02-07:
But are you sure it's really an error linked to this? 

On my side, the error appears instantly, the download hasn't even started

XECDesign, 2022-02-07:
The extension thing is just speculation on my end, I haven't checked how it determines the format to use.

It's linked in that trying to install gladys through rpi-imager doesn't work for me, but seemingly for different reasons. Are you running Imager 1.7.1?

Pierre-Gilles, 2022-02-07:
> Are you running Imager 1.7.1?

Yes !

Pierre-Gilles, 2022-02-07:
I wrote a very simple C++ program to try and see if the error is coming from curl download: 

```c++
#include <stdio.h>
#include <curl/curl.h>

int main(void)
{
    CURL *curl;
    FILE *fp;
    CURLcode res;
    char *url = "https://gladysassistant.com/download/latest";
    char outfilename[FILENAME_MAX] = "test.zip";
    curl = curl_easy_init();                                                                                                                                                                                                                                                           
    if (curl)
    {   
        fp = fopen(outfilename,"wb");
        curl_easy_setopt(curl, CURLOPT_URL, url);
        curl_easy_setopt(curl, CURLOPT_WRITEFUNCTION, NULL);
        curl_easy_setopt(curl, CURLOPT_WRITEDATA, fp);
        curl_easy_setopt(curl, CURLOPT_NOPROGRESS, 0);
        curl_easy_setopt(curl, CURLOPT_FOLLOWLOCATION, 1);
        curl_easy_setopt(curl, CURLOPT_MAXREDIRS, 10);
        curl_easy_setopt(curl, CURLOPT_FAILONERROR, 1);
        curl_easy_setopt(curl, CURLOPT_CONNECTTIMEOUT, 30);
        curl_easy_setopt(curl, CURLOPT_LOW_SPEED_TIME, 60);
        curl_easy_setopt(curl, CURLOPT_LOW_SPEED_LIMIT, 100);
        res = curl_easy_perform(curl);
        curl_easy_cleanup(curl);
        fclose(fp);
    }   
    return 0;
}
```

And it works fine, redirects are followed and the final file is a .zip file with the correct size.

lurch, 2022-02-07:
> there is a redirect (we use a generic URL that redirects to the latest version).

Is there much point in doing that, given that https://gladysassistant.com/img/external/raspberry-pi-imager/raspberry-pi-imager-gladys-assistant.json _also_ specifies `extract_size`, `extract_sha256`, `image_download_size`, `release_date` and `image_download_sha256` ? I guess if you update your "latest" link to redirect to a newer version, it means RPi Imager downloads of Gladys Assistant will be "broken" (hash-mismatch) until you also update those fields in `raspberry-pi-imager-gladys-assistant.json`, so it might make sense to just hardcode the exact image-URL anyway? :thinking:

Pierre-Gilles, 2022-02-07:
> Is there much point in doing that

We also use the redirection to redirect the user to the correct CDN depending on the user location (Github downloads are super slow in europe/asia, and not in the US, so we use a specific CDN to host the zip in Europe/Asia)

But I agree, for the "versioning" part, we could use a fixed URL here. There will still be a redirect for the CDN part

maxnet, 2022-02-07:
You are focussing too much in the redirect aspect.

>https://gladysassistant.com/download/latest

If you would like your image to be uncompressed try if it works better if you name the link latest.zip

Pierre-Gilles, 2022-02-07:
I confirm that changing the URL from `https://gladysassistant.com/download/latest` to `https://gladysassistant.com/download/latest.zip` fixed the issue ! 

[image]

Thank you very much for your help @maxnet & @XECDesign 🙏