raspberrypi/rpi-imager #334: [FEAT] Make the advanced options more accessible.

theLMGN, 2022-01-11:
I recently heard about the advanced features in passing, and when I went to go image an SD card, I decided that these advanced options would save a lot of time. I then downloaded the imager tool and spent a solid minute clicking the options to find this menu.

After Googling I found [this](https://www.raspberrypi.com/documentation/computers/getting-started.html#using-raspberry-pi-imager) page which ten told me I had to press Ctrl+Shift+X, which isn't obvious.

I feel that this is bad UX, as most people won't be aware of this keyboard shortcuts, even if it would be helpful to them!

I ask for a button somewhere on the main UI (such as another button in the button row or somewhere else)

lurch, 2022-01-12:
Duplicate of #180 ?

JasperAlgra, 2022-01-19:
Yes, second this! 

I just found out via a random tutorial... no way of knowing this feature even exists.
And when using it the second time after a while I forgot the "magical" key combination.. so I needed to spend some time again searching to find it. 

I would be very nice to have an "advanced" button. Maybe a bit hidden or bind a "are you sure?" modal if you must.

guysoft, 2022-01-19:
Hey,
If you want this perhaps you can tell me what you think about an unofficial fork, it seems like its what Rpi foundation suggests there:
https://github.com/raspberrypi/rpi-imager/issues/180#issuecomment-816451376

theLMGN, 2022-01-19:
> #180 (comment)

The reason I raise this issue is because it is inaccessible to those that
don't previously know of these settings. Making a fork does NOT improve
visibility.

While I'm sure using a terminal to set up WiFi is the "correct" way, for
most users clicking a box in an application on a computer that already
knows the password is easier, especially when they might not have a monitor
& keyboard to attach to the Pi.

guysoft, 2022-01-19:
You have a point regarding visibility. Unless the fork becomes more common knowledge.
If there are more community OSes onboard that use such a fork, it might make it more widely known. Since it seems keeping the UI simple makes sense if you want to be educational. Rpi Foundation people wrote in the issue there:
[#180 (comment)](https://github.com/raspberrypi/rpi-imager/issues/180#issuecomment-816451376) they don't want to do is.
It means there might be enough reason to make a community fork with more advanced options in general.

For example, you could add custom fields as a feature in the advanced tab (domain names, static IPs, you name it, more images to pick from).

maxnet, 2022-01-19:
There is code in git that will display an extra button (cog icon) if the user selected an image that advertises support for advanced settings.

To advertise support add to the images in your json the option: 

```
"init_format": "systemd" 
```

(or "init_format": "cloudinit" if your image is based on Ubuntu Server)

guysoft, 2022-01-19:
@maxnet Cool, I understood that is in the works (from our talk here https://github.com/raspberrypi/rpi-imager/issues) .

maxnet, 2022-01-19:
>Cool, I understood that is in the works

Just no distributions in the repository that advertise support right now.
But since you are maintaining your own .json file, you do can add the option to make it work for your distro...

guysoft, 2022-01-19:
I don't have an Ubuntu-based distro on the official JSON atm.
Also my Ubuntu-based distros can also handle wifi settings from the raspbian distro format thanks to [CustomPiOS network module](https://github.com/guysoft/CustomPiOS/wiki/Modules#network). So if you pick them as a custom images advanced settings for wifi do work.

maxnet, 2022-01-19:
>I don't have an Ubuntu-based distro on the official JSON atm.

If you have RPI OS based distros the setting would be   "init_format": "systemd" 

(instead of cloudinit)

cillian64, 2023-10-19:
OS customisation is now accessible without a keyboard shortcut (and Ubuntu server is in the standard OS list with customisation enabled)

closed 2023-10-19 by cillian64