raspberrypi/rpi-imager #336: [Feature Request] Ability to change user password even if using key-based auth

jpsingleton, 2022-01-18:
I usually use key-based auth but it would be nice to change the default password of the pi (or other) user too even when using this, to harden security without having to login and run `passwd`. May be related: https://github.com/raspberrypi/rpi-imager/issues/287 (looks like this is done but not released?).

maxnet, 2022-01-18:
>I usually use key-based auth but it would be nice to change the default password of the pi (or other) user too even when using 
>this, to harden security without having to login and run passwd. 

Note that when you select key-based auth it disables the option to login through SSH by password completely.
Which is best security practice.

If we allow setting a password as well, then it would look to the user that he could use that password for ssh as well.

jpsingleton, 2022-01-19:
Yeah, the UI might need rejigging. SSH by password should be disabled but I'm thinking of local access. Leave the pi user with the default password may be a risk for some users and they won't realise.

Maybe changing the default password should be a separate option and required if SSH is enabled? Bonus points if it generates a secure unique random password for the user too. Or perhaps just some text saying to change the password if key-based auth is used?

maxnet, 2022-01-19:
>but I'm thinking of local access. Leave the pi user with the default password may be a risk for some users and they won't realise.

Well, auto-login is enabled by default locally, and sudo does not require password either.
No matter what password is set, locally there is no security whatsoever.

closed 2022-01-19 by jpsingleton