raspberrypi/rpi-imager #337: [Feature Request] Please add Volumio

rexjohannes, 2022-01-18:
Please add Volumio RPI Image (https://volumio.com/en/get-started/) to the Repo.

ghollingworth, 2022-01-18:
Do you have a suitable json file for it?

rexjohannes, 2022-01-18:
Yeah

```
{
  "os_list": [
    {
        "name": "Volumio",
        "description": "Volumio Audiophile Music Player for Raspberry Pi",
        "icon": "https://cdn.volumio.com/wp-content/uploads/2021/08/logo-circle-grey-transparent-e1585162923840-150x150.png",
        "url": "https://updates.volumio.org/pi/volumio/3.179/Volumio-3.179-2021-12-20-pi.zip",
        "extract_size": 2946498560,
        "extract_sha256": "5E4053505AB6069BCB348D590781DC8676BD18BE15CD480E1FF41DB754C36B55",
        "image_download_size": 440631296,
        "release_date": "2021-12-20"
    }
  ]
}
```

ghollingworth, 2022-01-18:
No you need to store that on your server and provide me with something looking like this

                {
                    "name": "OSMC",
                    "description": "A fast and feature filled open source media center",
                    "website": "https://osmc.tv",
                    "icon": "http://download.osmc.tv/installers/noobs/NOOBS-Logo-2.png",
                    "subitems_url": "http://download.osmc.tv/installers/rpidiskwriterimages.json"
                }

rexjohannes, 2022-01-18:
Oh okay, then here:

```
            {
                "name": "Volumio",
                "description": "Volumio Audiophile Music Player for Raspberry Pi",
                "website": "https://volumio.com",
                "icon": "https://cdn.volumio.com/wp-content/uploads/2021/08/logo-circle-grey-transparent-e1585162923840-150x150.png",
                "subitems_url": "http://download.osmc.tv/installers/rpidiskwriterimages.json"
            }
```

ghollingworth, 2022-01-18:
You do realise the important bit of that is the bit you didn't change, i.e. the pointer to the subitems_url?

What you (I assume a Volumio developer) need to do:

1) Write a subitems json file to list the information in the image(s) you have on your site
2) publish that file on your site as a https link (http://download.osmc.tv/installers/rpidiskwriterimages.json is OMSC's .json file you could read this for comparison)
3) Write the entry to go into the Raspberry Pi top-level .json file (as above) the top level is here: http://downloads.raspberrypi.org/os_list_imagingutility_v2.json)
3) Test the top-level .json by downloading it locally editing it to add your section with your subitems URL and then running rpi-imager --repo top-level.json
4) submit the section to us

Note, we can only accept these from the Volumio developer in control of the website

Gordon

closed 2022-01-18 by rexjohannes