raspberrypi/rpi-imager #342: Allow SSH public_key as well as password

debuglevel, 2022-01-30:
In 1.6.2, one has to choose between a SSH password or a SSH public key. It would be nice if a combination of both would also be possible.

lurch, 2022-01-31:
Duplicate of #336 ?

maxnet, 2022-01-31:
>Duplicate of #336 ?

Think 336 was more about setting a password for other purposes (like local logins).
And that is possible since 83513733ef0963ef7ea64c9efeec84479f97639a

Not about allowing less secure SSH password access, while the user did set a proper SSH key.

CharlesGodwin, 2022-04-10:
I resolved this with this bash snippet. The test is so it can be run more than once without making a mess of the file
The code enables password authentication. The imager disables it by default, which I think is a good thing for security purposes.

I run this script after my initial SSH *with key" login.

<pre>if ! grep -q "^PasswordAuthentication yes" /etc/ssh/sshd_config; then
    sudo sed -i s/^PasswordAuthentication/\#PasswordAuthentication/ /etc/ssh/sshd_config
    echo "PasswordAuthentication yes"|sudo tee -a /etc/ssh/sshd_config >/dev/null
fi</pre>

tdewey-rpi, 2024-02-06:
Not clear there's a concrete action to be taken here, nor that one would be desirable, nor what it would be - as #444 appears to directly contradict this issue.

Closing as stale, please raise a new issue if you're still seeing ssh authentication behaviour you consider wrong.

closed 2024-02-06 by tdewey-rpi