raspberrypi/rpi-imager #343: Advanced Options Menu not available in v1.7

mcguirepr89, 2022-02-03:
Hello and thank you for `rpi-imager`

I just rebuilt to upgrade to version 1.7 and no longer have access to the "Ctrl+Shift+x" Advanced Options page.
My OS= Debian Bullseye x86_64

When using those key presses, the console prints the following (I preserved the starting console output as well):
```
patrick$: rpi-imager https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/2022-01-28-raspios-bullseye-arm64-lite.zip
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qrc:/main.qml:304:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
```
I've confirmed that the same thing happens when using an image from the online Operating System list. The output below is from selecting the default RaspiOS 32-bit and trying to click the Settings icon and trying to press "Ctrl+Shift+X"
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qrc:/main.qml:304:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
QObject::setParent: Cannot set parent, new parent is in a different thread
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool
```

Thank you in advance for your help with this,
Patrick

jssalter, 2022-02-03:
Also having this issue. There's now a cog that shows up in the window which, presumably, is to the advanced settings, but nothing happens when I click on it. Trying to find a way to roll back to 1.6

mcguirepr89, 2022-02-03:
@jssalter You should be able to downgrade your Windows version by simply grabbing a 1.6.exe [from here](https://downloads.raspberrypi.org/imager/) -- I hope that is helpful for you.

lurch, 2022-02-03:
@jssalter https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7 says the advanced settings option only works if RPi Imager detects that the image is "configurable" ?

mcguirepr89, 2022-02-03:
@lurch I've confirmed that this is happening for the default RaspiOS Desktop 32-bit.

ghollingworth, 2022-02-03:
Can you confirm what OS you have reproduced this on?

Did you run rpi-imager on Raspberry Pi OS?

geerlingguy, 2022-02-03:
I'm running it on macOS Big Sur (11.6) and there's no options that appear when I press Shift + Ctrl + X anymore.

ednl, 2022-02-03:
Same for me on MacOS Big Sur 11.6.3 with the RasPiOS 64-bit Lite image. A gear icon appears but does nothing. Ah, as I was typing Jeff uploaded a gif; so yeah, that.

mcguirepr89, 2022-02-03:
As mentioned when creating the issue, I'm on Debian Bullseye x86_64

aallan, 2022-02-03:
Works for me on macOS Catalina. It may be a permissions thing on Bug Sur, because it **does** prompt me before opening the window asking permission to give the Wi-Fi password from my keychain.

[image]
 
Permissions get a lot more complicated on macOS post-Catalina.

ednl, 2022-02-03:
@aallan Thanks. It may be that but the previous version 1.6.2 of Imager works well on Big Sur and Monterey. And the OP has this issue on Debian x86.

maxnet, 2022-02-03:
>qrc:/OptionsPopup.qml:488: Error: Cannot assign [undefined] to bool

Ah, caused by the hidden wifi ssid code.
That incorrectly assumes that if there are stored wifi settings, there is also always a setting stored that says whether or not the wifi ssid is hidden.
However users that were using ctrl+shift+x in a previous version, saved wifi settings, and chose to use save settings "to always use" do not have that setting...

As a workaround for now, deleting your stored settings should work.

Linux:

```
rm  ~/.config/Raspberry\ Pi/Imager.conf
```

Mac OSX:

```
rm ~/Library/Preferences/org.raspberrypi.Imager.plist
```

geerlingguy, 2022-02-03:
@maxnet - After quitting, deleting the preferences file, and opening Imager again, I see no difference (keyboard shortcut doesn't work before or after selecting an image, settings cog never does anything when clicked).

geerlingguy, 2022-02-03:
The plot thickens—attaching a video because it's the easiest way to illustrate, but basically, when I choose to 'Write', the only way in 1.7 for me to actually be able to write an image is to clear all settings—the settings choice dialog appears but again, clicking on any of the options to configure the advanced options results in nothing happening:

https://user-images.githubusercontent.com/481677/152397752-a8fbc2e5-093f-41c1-bcf4-b64c7157c7a3.mov

aallan, 2022-02-03:
Just tested it on another laptop, this one running macOS Big Sur 11.6.1. It works for me.

I get the same prompt to pass wifi credentials from my keychain, but either way the Advanced Options menu **does** appear when I hit the cog wheel.

maxnet, 2022-02-03:
>@maxnet - After quitting, deleting the preferences file, and opening Imager again, I see no difference

Strange.

Can you confirm or deny that these builds do fix the problems?

[Raspberry Pi Imager 1.7.1.dmg](https://www.maxnet.eu/tmp/171-hiddenssid/Raspberry%20Pi%20Imager%201.7.1.dmg)
[rpi-imager_1.7.1_amd64.deb](https://www.maxnet.eu/tmp/171-hiddenssid/rpi-imager_1.7.1_amd64.deb)

mcguirepr89, 2022-02-03:
Hello, @maxnet -- thanks for your great fix. I just built from https://github.com/raspberrypi/rpi-imager/commit/d91f8d95b1c9eab3153d0a6f32abeaeb95f87104 and can confirm that things are working again. 

THANK YOU!!!

meteyou, 2022-02-03:
@maxnet in v1.7.0 i had exactly the same bug (macOS monterey 12.0.1) but with v1.7.1 it's fixed for me as well

geerlingguy, 2022-02-03:
Everything's working as expected on 1.7.1.

maxnet, 2022-02-03:
Good to hear things are fixed in 1.7.1
Now also on the release page. Closing this.

closed 2022-02-03 by maxnet

ednl, 2022-02-03:
Yes, 1.7.1 works on Big Sur, this is without having deleted my old "always use" settings. Thanks!

jssalter, 2022-02-03:
> @jssalter https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7 says the advanced settings option only works if RPi Imager detects that the image is "configurable" ?

I was trying to use Raspbian 32-bit.

> @jssalter You should be able to downgrade your Windows version by simply grabbing a 1.6.exe [from here](https://downloads.raspberrypi.org/imager/) -- I hope that is helpful for you.

Thank you

mcguirepr89, 2022-02-03:
I'm afraid I need to revisit this issue, as the appearance and setting of the Advanced Options menu is not actually writing the `firstrun.sh` script.
Please see the output below of an attempt to set these things, noting the empty `Custom cmdline.txt entries: ""`, which I believe is what directs the Pi to the `firstrun.sh` script.
```
patrick @ bridnetpidemo ➜  ~  rpi-imager
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qrc:/main.qml:304:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
QObject::setParent: Cannot set parent, new parent is in a different thread
Custom config.txt entries: ""
Custom cmdline.txt entries: ""
Custom firstuse.sh: "#!/bin/bash\n\nset +e\n\nFIRSTUSER=`getent passwd 1000 | cut -d: -f1`\nFIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`\nif [ -f /usr/lib/userconf-pi/userconf ]; then\n   /usr/lib/userconf-pi/userconf 'pi' '$5$mXq2ewJlEK$2tGL.tc/O6iSkZgjer03Dw5wsvRSuwoARXGSvG24nL2'\nelse\n   echo \"$FIRSTUSER:\"'$5$mXq2ewJlEK$2tGL.tc/O6iSkZgjer03Dw5wsvRSuwoARXGSvG24nL2' | chpasswd -e\n   if [ \"$FIRSTUSER\" != \"pi\" ]; then\n      usermod -l \"pi\" \"$FIRSTUSER\"\n      usermod -m -d \"/home/pi\" \"pi\"\n      groupmod -n \"pi\" \"$FIRSTUSER\"\n      if grep -q \"^autologin-user=\" /etc/lightdm/lightdm.conf ; then\n         sed /etc/lightdm/lightdm.conf -i -e \"s/^autologin-user=.*/autologin-user=pi/\"\n      fi\n      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then\n         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e \"s/$FIRSTUSER/pi/\"\n      fi\n      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then\n         sed -i \"s/^$FIRSTUSER /pi /\" /etc/sudoers.d/010_pi-nopasswd\n      fi\n   fi\nfi\nsystemctl enable ssh\ncat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'\ncountry=US\nctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev\nap_scan=1\n\nupdate_config=1\nnetwork={\n\tssid=\"Fast Magnolia\"\n\tpsk=cfd0a675ba04482b7ef5e03c6556885a6bd3b477bd72de685ec07a93d6e4848f\n}\n\nWPAEOF\nchmod 600 /etc/wpa_supplicant/wpa_supplicant.conf\nrfkill unblock wifi\nfor filename in /var/lib/systemd/rfkill/*:wlan ; do\n  echo 0 > $filename\ndone\nrm -f /etc/xdg/autostart/piwiz.desktop\nrm -f /etc/localtime\necho \"America/New_York\" >/etc/timezone\ndpkg-reconfigure -f noninteractive tzdata\ncat >/etc/default/keyboard <<'KBEOF'\nXKBMODEL=\"pc105\"\nXKBLAYOUT=\"us\"\nXKBVARIANT=\"\"\nXKBOPTIONS=\"\"\n\nKBEOF\ndpkg-reconfigure -f noninteractive keyboard-configuration\nrm -f /boot/firstrun.sh\nsed -i 's| systemd.run.*||g' /boot/cmdline.txt\nexit 0\n"
Cloudinit: "users:\n- name: pi\n  groups: users,adm,dialout,audio,netdev,video,plugdev,cdrom,games,input,gpio,spi,i2c,render,sudo\n  shell: /bin/bash\n  lock_passwd: false\n  passwd: $5$mXq2ewJlEK$2tGL.tc/O6iSkZgjer03Dw5wsvRSuwoARXGSvG24nL2\n\nssh_pwauth: true\n\ntimezone: America/New_York\nruncmd:\n- sed -i 's/^s*REGDOMAIN=S*/REGDOMAIN=US/' /etc/default/crda || true\n- rm -f /etc/xdg/autostart/piwiz.desktop\n- localectl set-x11-keymap \"us\" pc105\n- setupcon -k --force || true\n\n\n"
```
I have mounted the image after attempting to write with these customizations and have confirmed that there is no `firstrun.sh` script in `boot`.

Thank you again for your attention to this,
Patrick

maxnet, 2022-02-03:
>I'm afraid I need to revisit this issue, as the appearance and setting of the Advanced Options menu is not actually writing the 
>firstrun.sh script.

Using what distribution?
Note that distributions must now advertise wheter they support systemd (firstrun.sh) or cloud-init customization for things to work properly.

>Please see the output below of an attempt to set these things, noting the empty Custom cmdline.txt entries: ""

The code to modify cmdline.txt if necessary is now at a different place.
At the stage that is printed it does not know yet what customization method it should apply

mcguirepr89, 2022-02-03:
As mentioned when I opened the issue, I'm on Debian Bullseye x86_64

The distro I was trying to write to was RaspiOS Lite ARM64 found [here](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/)

I will assume, then, that the 64-bit version of RaspiOS will not accept the ability to run a `firstrun.sh` script, which I admit seems wrong. I will continue to use 1.6.2 since that version _did indeed_ write to RaspiOS Lite ARM64.

maxnet, 2022-02-03:
>The distro I was trying to write to was RaspiOS Lite ARM64 found [here](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-01-28/)

That one definitely does work.
If you bring your own image ("use custom"), it does apply firstrun.sh customization if there is an issue.txt on the FAT partition and it contains "pi-gen"

Did you check for firstrun.sh immediately after write, or did you boot the SD card first?
As on first boot the Pi does remove firstrun.sh after consuming it.

mcguirepr89, 2022-02-03:
I checked for `firstrun.sh` in `/boot` just after writing (it is not there). The `issue.txt` file does contain the following:
```
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage2
```
I'm not sure what else to look for.

maxnet, 2022-02-03:
[image]

Am afraid I cannot reproduce your problem.
Alters cmdline.txt fine and adds firstrun.sh with Imager 1.7.1 running under Kubuntu with that image.

mcguirepr89, 2022-02-03:
Thank you very much for your work -- there must be something wrong on my end. I appreciate your help

ednl, 2022-02-06:
No, I had the same issue with writing PiOS 64 Full from MacOS; no customisations were written, not even ssh was enabled. The image write was to a usb ssd, not an sd card, but I didn't think that would be different. The image worked, just no customisations.

maxnet, 2022-02-06:
>No, I had the same issue with writing PiOS 64 Full from MacOS; no customisations were written, not even ssh was enable

Does it concern settings saved in an earlier release to always use, with ssh public key authentication and did you use any beta release (such as the m1 build) that allowed you to change the username without setting password?

Then you may need to change settings, either enter a password or uncheck the set "username and password" box, and "save settings".
Make sure the settings box closes after pressing "save" or one of your settings may be wrong. Scroll up and down to see of anything is marked in red.

Problem described here: https://github.com/raspberrypi/rpi-imager/issues/345#issuecomment-1030652142

Obad94, 2023-12-29:
Same happened with me on raspberry pi imager version 1.8.4. There is no advanced setting option available.

MGX-CODING, 2024-04-12:
Just to add oil to the flames, I'm on Windows 11, using imager v1.8.5 and I have no way of opening advanced settings from the UI.

I saw someone above metioning Ctrl + Shift + X, which did the trick, but without that kind soul, I would have been stuck with my default, boring settings.