raspberrypi/rpi-imager #344: Imager 1.7 stores WiFi as plain text to its imager.conf

beta-tester, 2022-02-03:
i just realizes that `rpi-imager` is storing the password for WiFi as plain text to its `~/.config/Raspberry Pi/Imager.conf` file after hitting [Ctrl]+[Shift]+[X]

```
...
wifiCountry=DE
wifiPassword=password1234
wifiSSID=WLAN
wifiSSIDHidden=false
```
i would suggest to store it as `wpa_passphrase` like:

```
...
wifiCountry=DE
wifiPassword=00ef33a07b61ca4bb0eebed7238fc0b260570ea474f15460cf3b5aaa7660c067
wifiSSID=WLAN
wifiSSIDHidden=false
```

maxnet, 2022-02-03:
Problem is that we do not know at that point yet, if the operating system you are going to install supports that.

The customization method we use for RPI OS does supports that. And we do write the wifi password encoded as PBKDF2 hash to firstrun.sh to SD card in that case.

But it could also be that you want to install Ubuntu server, and I don't think cloud-init/netplans supports that.
At least I couldn't find it in the fine documentation: https://netplan.io/reference/#properties-for-device-type-wifis%3A

beta-tester, 2022-02-03:
oh, i thought those settings are only for RasPi OS and there it is known and wpa_passphrase is at every distro the same.
ok, if not, then it is indeed really tricky.

maxnet, 2022-02-03:
Problem is that while Ubuntu uses wpa_supplicant under the hood as well, we cannot write that file directly.
If we do, our changes will be overwritten by Netplan on the next boot, that manages the file...

beta-tester, 2022-02-04:
strange that there were no security complains about that "_ubuntu_" / netplan store passwords in plain text.
would they change it when a i complains?

maxnet, 2022-02-04:
Can always try.
But I think they have bigger things to fix first.
Like that they do not offer a clean way to set wifi country code either, so 5 ghz and EU channels not working properly.

ghollingworth, 2022-02-04:
I would think it better to store it in an encrypted format and then only support that format in cloud-init for the moment.  I'm sure Canonical will be able to fix up their server builds to work with it...  

It's definitely the 'right' way to do it no matter what is currently implemented...

beta-tester, 2022-02-04:
will see what will happen...
https://bugs.launchpad.net/netplan/+bug/1960015

closed 2022-02-04 by maxnet

beta-tester, 2022-02-04:
did you saw the answer of the bug report?
it seems it is already implemented in netplan but wasn't documented yet.

@maxnet  i see you changed already some code.
very nice... thank you.

commit 4bd44634 (2022-02-04): Store saved wifi password hashed

Does have the consequence that if the user modifies the SSID, he
will also need to enter the password again, as PKDF2 scheme hashes
SSID together with password.

Closes #344
commit 530e8977 (2022-02-04): Store saved wifi password hashed

Does have the consequence that if the user modifies the SSID, they
will also need to enter the password again, as PKDF2 scheme hashes
SSID together with password.

Closes #344
commit e01c64df (2022-02-04): Store saved wifi password hashed

Does have the consequence that if the user modifies the SSID, they
will also need to enter the password again, as PBKDF2 scheme hashes
SSID together with password.

Closes #344
commit eb011bb0 (2022-02-04): Store saved wifi password hashed

Does have the consequence that if the user modifies the SSID, they
will also need to enter the password again, as PBKDF2 scheme hashes
SSID together with password.

Closes #344