raspberrypi/rpi-imager #345: v1.7.1: Wifi settings & Locale not configured properly w/advanced options(ctrl shift x)

Josmopolitan, 2022-02-04:
Tested on Zero 2 W:

Generated an image for Retropie with a wifi country of 'US' and keyboard layout of 'us' and my home network settings. Upon booting the pi there is no network connectivity. Going into wifi settings in retropie shows wlan0 as undefined and entering raspberry pi config generates a message of no wifi country configured. Additionally, the locale and keyboard is set to en-GB. I tested with multiple images written from 2 different computers with the same results on 2 different raspberry pi zero 2 W boards.

Additionally, I tested an octopi image and got similar results. However, inside the boot directory there were 2 files for wpa_supplicant and a wifi settings txt file. They reflected a GB wifi country and the wifi settings were blank. I manually entered my information and had success with the octopi image, after correcting the incorrect files in the boot directory.

I was unable to find similar files in the retropie boot directory.

maxnet, 2022-02-04:
Retropie does not advertise supporting customization. So that no settings are applied is to be expected.

Octopi does.
@guysoft does your script that copies settings also handles things like locale or only a subset of options?

guysoft, 2022-02-04:
Hey, just tested on v1.7.1 (fresh download from today) with OctoPi 0.18.0, both hostname, password and wifi were set up correctly.
The format (removed my wifi settings):
```
country=IL
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

update_config=1
network={
        ssid="MY_WIFI"
        psk=LOTS_OF_HEX
}
```

The file ``/boot/octopi-wpa-supplicant.txt`` was edited and has the wifi settings.

Also ``/etc/wpa_supplicant/wpa_supplicant.conf`` is set up right as a symlink to ``/boot/octopi-wpa-supplicant.txt``.

I could ssh and all worked.

Josmopolitan, 2022-02-04:
Okay, I tested Octopi again and I was mistaken and the pi did boot up and dhcp kicked in with the wifi settings. However, the octopio-wpa-supplicant.txt does default to having the GB country uncommented, and I don't know if that matters.

Retropie, however, there is no network interface at all that generates on a new image.

guysoft, 2022-02-04:
Indeed, we did that at the time Rpi 3 came out.
Initially I though of adding all countries, but we voted against it:
https://github.com/guysoft/CustomPiOS/commit/d601dd15bb2a0a601afb87770d21daee6640d533

ghollingworth, 2022-02-04:
Retropie doesn’t support systemd or cloudinit configuration, that’s why the button doesn’t appear when you select it

maxnet, 2022-02-04:
>Initially I though of adding all countries, but we voted against it

Suggests you do allow all countries as wifi country setting, or it may be sending on frequencies it is not supposed to (even if you do not have an access point there, it still sends when doing active scans for available APs)
Channel 14 may be normal to use in say Europe, it belongs to the military elsewhere.

guysoft, 2022-02-04:
Retropie images are built according to what I was told here in 2020 by building on top of Raspbian Lite: https://retropie.org.uk/forum/topic/25676/how-do-you-build-the-retropie-images-that-you-ship/3?_=1643998346978
Not sure why it wont work, just thought I'd like that information here. It was a challenge to find it at the time.

guysoft, 2022-02-04:
@maxnet Is there a country who's channels are ok in all other counties?
Because otherwise its going to cause a mess.
cc @foosel

maxnet, 2022-02-04:
>@maxnet Is there a country who's channels are ok in all other counties?

Think there used to be in the 2.4 Ghz era. But with 5 Ghz things are more complicated.

https://en.wikipedia.org/wiki/List_of_WLAN_channels

Josmopolitan, 2022-02-04:
Not sure if it's helpful or not, but I was able to install retropie on a 3B last week without any after-the-fact configuration necessary. it worked out of the box and everything functions as expected.

Josmopolitan, 2022-02-04:
And the main issue I'm having it it seems VERY difficult to fix after the fact. If you try to search for wifi in retropie, it says there's no country set. If you try to set the country in raspi-config, it says the wireless device, wlan0, does not exist. And as an entry to mid level user and going through and manually setting the language, locale, etc, as well as manually entering an /etc/network/interfaces file and a wpa_supplicant.conf, nothing seems to allow you to initiate network services or get the OS to recognize the presence of a network device. It might be an issue unique to the Zero 2 W, since it seems to have worked fine on the 3B, but I can't say for sure, I tried imaging using both the RPI 2/3 image as well as the RPI 1/Zero image and no dice with either. If I could "fix" it after the fact, it wouldn't be an issue. The only thing I haven't tried is imaging to 100% total defaults and then trying to go in and set it up completely manually, but there's a disconnect between what Raspberry Pi Imager maybe can or can't do--or a specific issue with the Zero 2.

maxnet, 2022-02-04:
>Not sure if it's helpful or not, but I was able to install retropie on a 3B last week without any after-the-fact configuration necessary.

The previous Imager version indeed allowed settings to be applied to any image.
The downside of that was that with some distributions that did not work (for example because the image did not mount the FAT partition at /boot but somewhere else), and those could hang on boot.
So we now only apply the settings, if the distribution has indicated it supports the customization we do.

@RetroPie
@joolswills
If RetroPie is based on RPI OS, and supports the way Imager applies advanced settings (which Josmoplotian indicates it does), can you please add to your entries in https://retropie.org.uk/os_list_imagingutility.json the following attribute:

```
"init_format": "systemd"
```

joolswills, 2022-02-04:
I will test this for the next image.

guysoft, 2022-02-04:
@maxnet Reguarding the country code set. There seems to be a 00 mode in all linux wifi devices. Managed to talk to OpenWRT guys to get an answer to that. you can set/get it with ``iw``.

You can set and get it with ``iw reg set`` and ``iw reg get``.
It includes 5ghz bands.

Perhaps that could be the default and in most cases it would give some kind of wifi functionality?

```
$ sudo   iw reg set 00
$ iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country 00: DFS-UNSET
        (2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
        (2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
        (2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
        (5170 - 5190 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
        (5190 - 5210 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
        (5210 - 5230 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
        (5230 - 5250 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
        (5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5490 - 5510 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5510 - 5530 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5530 - 5550 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5550 - 5570 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5570 - 5590 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5590 - 5610 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5610 - 5630 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
        (5630 - 5650 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
        (5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
        (5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
```

maxnet, 2022-02-04:
>@maxnet Reguarding the country code set. There seems to be a 00 mode in all linux wifi devices.

Ah, that is good to know.

Thinking of it, I think I did actually experiment with that in Berryboot some years ago.
Only used it during initial setup though.
There I instructed "iw" to perform a passive scan (that does not send stuff), and looked at the various country codes broadcasted by the various access points in the neighboorhoud.

As in:

```
max@pi99:~ $ sudo iw wlan0 scan passive |grep Country:
        Country: XA     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: XA     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: BE     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
        Country: NL     Environment: Indoor/Outdoor
        Country: DE     Environment: Indoor/Outdoor
        Country: DE     Environment: Indoor/Outdoor
        Country: EU     Environment: Indoor/Outdoor
```

I would then assume the country that is advertised the most is the correct region, and offer that as first choice in the GUI for the user to confirm, to prevent them scrolling through a big list.

hbarnstone, 2022-02-05:
Hewo!!❤️😘

maxnet, 2022-02-05:
>Attempt to SSH to the RPi, or see on your router if it connects to wifi, or check its hostname on your router

Have you tried connecting a display to the Pi, and checked even the hostname is not set there? (it is displayed at login prompt)

It ia very easy to make mistakes in wifi configuration. E.g. some forget that not only passwords but also SSIDs are CaSE SeNSitive.

williamcpierce, 2022-02-05:
> > Attempt to SSH to the RPi, or see on your router if it connects to wifi, or check its hostname on your router
> 
> Have you tried connecting a display to the Pi, and checked even the hostname is not set there? (it is displayed at login prompt)
> 
> It is very easy to make mistakes in wifi configuration. E.g. some forget that not only passwords but also SSIDs are CaSE SeNSitive.

Don’t have the right connectors so can’t test this way, but:
- Wi-Fi settings were not touched at any point, I specified them on version 1.6.2 when I set up a pi last week. They appeared the same after upgrading to 1.7.1, and worked again after downgrading to 1.6.2
- After manually adding wpa_supplicant.conf and the blank ssh file to my sd card, I was able to ssh in, and saw that the hostname was still the default “raspberrypi” rather than my setting. I don’t believe these files affect the hostname, but could be mistaken

maxnet, 2022-02-05:
>Don’t have the right connectors so can’t test this way

Another way to check is to open the "boot" volume on your SD card after writing and look if a firstrun.sh was created (or user-data)
May require replugging the SD card (or your USB reader), as Imager "ejects" the volume by default.
Do this before putting the card in the Pi, as once consumed on boot firstrun.sh will self destruct.

Not able to reproduce your issue (with RPI OS Lite from repository):

[image]

A change do is that in 1.7.1 Imager will only apply the customizations for OS images flagged as supporting customization.
But that is the case for the standard RPI OS (Lite) images from repository.

williamcpierce, 2022-02-05:
Tested it again with your suggestion to check the boot disk without putting it in the pi, and had the same result. 
1. Wrote an image using 1.6.2 and my custom settings, and cmdline.txt and firstrun.sh were created as expected
2. Upgraded to 1.7.1 by downloading from https://www.raspberrypi.com/software/ and dragging the new application to my Applications folder, and selecting "replace"
3. Checked settings in 1.7.1, they appeared identical to what I had set in 1.6.2
4. Wrote the image again, and firstrun.sh was NOT created. cmdline.txt was missing the line referencing it as well:
[image]
[image]

maxnet, 2022-02-05:
Does it better if you uncheck the new "Set username and password" option, press "save" and try again?

williamcpierce, 2022-02-05:
> Does it work better if you uncheck the new "Set username and password" option, press "save" and try again?
> 
> You do are using RPI OS lite from the built-in menu? If it is a saved .img file used with "use custom" instead, did it create a "user-data" file instead of "firstrun.sh"?

Ah there we go, unchecking "Set username and password" fixed the issue. Didn't notice it had changed between versions/assumed blank password would work if using public-key authentication, but it makes sense that it wouldn't. Thanks!

Was using RPI OS lite from the built-in menu.

Josmopolitan, 2022-02-05:
I have crossposted this issue to the Retropie forums as it appears the issue is persistent without using any customization options and may also be a problem unique to the Retropie image relating to the hardware configuration of the Zero 2 W. The wifi card simply doesn't exist on a fresh install of Retropie on a zero 2 w. 

https://retropie.org.uk/forum/topic/32167/install-from-image-on-pi-zero-2-w-disables-or-fails-to-install-drivers-firmware-wifi-card

bstrech, 2022-02-10:
> I'm having a similar issue, 1.7.1 is not applying any specified settings. Settings were still able to be set by manually adding files. Reverting to 1.6.2 also fixed the issue.
> 
> Device: Raspberry Pi Zero 2 W SD Card: SanDisk Edge 16gb OS: MacOS Monterey 12.1 Raspberry Pi Imager: 1.7.1 Raspberry Pi OS: Raspberry Pi OS Lite (32-bit) - Jan 28 2022 release Settings checked/modified:
> 
> * Set hostname
> * Enable SSH: Allow public-key authentication only
> * Configure wifi
> * Wifi country: US
> * Eject media when finished
> * Image customization options: for this session only
> 
> To reproduce:
> 
> 1. Write an image of RPi OS Lite (32-bit), with the above settings specified in RPi Imager 1.7.1
> 2. Boot off this image on a RPi
> 3. Attempt to SSH to the RPi, or see on your router if it connects to wifi, or check its hostname on your router
> 
> To resolve: Add wpa_supplicant.conf or ssh files manually to the boot disk, or revert to 1.6.2

I am also having this same behavior. 1.6.2 works, but 1.7.1 does not.
The hostname, ssh enable, and wpa_supplicant (wifi config) is not setup correctly. I can manually put a ssh file and wpa_supplicant.conf file on the SD card and that works. However, I cannot set the hostname and that is my main problem. I already have raspberrypi.local on my network.

williamcpierce, 2022-02-10:
@bstrech Check to see that you didn't make the same mistake I did, per https://github.com/raspberrypi/rpi-imager/issues/345#issuecomment-1030652142 and the following comments. 
1.7.1 added a new option to "Set username and password", but you need to supply both a username and password if you have that box checked. Otherwise the configuration fails silently and the image is copied but without any custom settings

bckelley, 2023-11-29:
I also tried using an image from debian and advanced settings (v1.8.3) and still didn't work.

maxnet, 2023-11-29:
>I also tried using an image from debian

Please be very specific about the exact image you used.

Did you chose RPI OS from the menu?

Or did you download upstream Debian from debian.org ?
We do not support customizing the latter.

abe-winter, 2023-11-29:
Hi, I'm seeing similar behavior on imager 1.8.1 and 1.8.3 where wpa_supplicant.conf doesn't have requested settings.

I didn't test older versions, but I can if that's helpful.

My wpa_supplicant.conf (no ssid / passphrase):
```
$ sudo cat etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```

1.8.1 setup:
[image]

1.8.3 setup: 
[image]

customization screen (from hitting ctrl-shift-X):
[image]

happy to help debug this in any way, or do more checks to make sure this isn't an error on my side. LMK + thanks.

tdewey-rpi, 2024-02-06:
Closing as won't fix.

This issue has been redirected a few times, which makes it hard to determine what actions, if any, needed to occur here. From reading the issue, it looks like people have largely found success or have stopped responding. I note that the 1.8.x series included a number of critical correctness fixes, so any new issue raised should be tested against the latest release.

closed 2024-02-06 by tdewey-rpi