raspberrypi/rpi-imager #350: Creating am Appimage for rpi-imager

abhas, 2022-02-05:
I would like to help with creating an updated Appimage for the rpi-image so that it could be used on any GNU/Linux OS with ease and can be recommended as the general purpose image writer for users.

There is already an rpi-image Appimage available via the download site but its very old. I would be glad to help out if the original yml definition of the Appimage is still available somewhere.

Thanks for this lovely utility!

guysoft, 2022-02-06:
Same here, there isn't any documentation how to build it.

lurch, 2022-02-06:
There's some (older) info about the old AppImage in #20

maxnet, 2022-02-06:
I personally think it would be better if those other Linux distributions did some effort themselves and provided native packages.

Also keep in mind that Imager depends on other system services (such as UDisks2) to do its job. And you cannot stick those other services in an AppImage....
So that Imager would magically work on any Linux OS in existence if you would put Imager in a container is a bit too optimistic.

XECDesign, 2022-02-07:
I don't remember the specifics, but I found that the AppImage build was more trouble than it was worth. Also, it was downloaded only a handful of times.

guysoft, 2022-03-07:
Hey all, I managed to build an AppImage for my rpi-imager fork. 
You are welcome to test it and tell me if this works for you :)

Due to #200 it wont run on versions of Linux older than Ubuntu 20.04 LTS.

https://github.com/guysoft/pi-imager/releases/tag/v.1.7.1

maxnet, 2022-03-07:
>Due to https://github.com/raspberrypi/rpi-imager/issues/200 it wont run on versions of Linux older than Ubuntu 20.04 LTS.

Eh, why?

You did not stick a newer Qt version that does have that method in your AppImage?!

guysoft, 2022-03-07:
Because you need to build rpi-imager on older versions, and it wont build on anything older, see: https://github.com/probonopd/linuxdeployqt#a-note-on-binary-compatibility

maxnet, 2022-03-07:
>Because you need to build rpi-imager on older versions, and it wont build on anything older

Not with Ubuntu's 5 year old Qt packages, but recall you can get a recent Qt version for Ubuntu 18 from the Qt website.

guysoft, 2022-03-07:
Possibly. It might work. But at this point I have an AppImage and that's good enough for me.

SharkWipf, 2022-06-25:
> Also, it was downloaded only a handful of times.

Wouldn't that make sense, given it was, afaik, not listed on the main download page?  
Plus it was only "recent" for 3 months, after that newer versions came out that were not released as AppImage.
Not to mention awareness of AppImages has grown a lot since 2020, it was the first thing I looked for trying to get rpi-imager for my distro.  
Especially considering the only official build instructions available are "how to build a deb", other distros can't even realistically package their own unofficial versions of rpi-imager, so having a distro-agnostic official AppImage makes perfect sense to me, more sense even than having an Ubuntu-native version.

lurch, 2022-09-22:
> I'll find another way to update the bootloader.

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#rpi-eeprom-update
https://github.com/raspberrypi/rpi-eeprom/

maxnet, 2022-09-22:
>I can't install the package available for download on the main page (for Ubuntu x86, how ridiculous) on debian.

Then following the instructions on the main page to build the package yourself instead.

https://github.com/raspberrypi/rpi-imager

it9exm, 2022-09-22:
Found [here](https://github.com/raspberrypi/rpi-eeprom/releases) what I was looking for.
And no, I won't build and install that stuff on my system. I would have gladly given a try if that came the form of an appimage, like the most sane software projects. Now I just don't need that any longer.

TheAssassin, 2022-10-15:
Started working on some AppImage build in my [fork](https://github.com/TheAssassin/rpi-imager). Provides some AppImages for testing in https://github.com/TheAssassin/rpi-imager/releases/continuous. More information can be found in #495. Hopefully, the developers reconsider shipping AppImages, as they're by far more suitable than native packages for upstream packaging, and easier to use for users as well.

guysoft, 2022-10-18:
Cool
@TheAssassin  is there any advantage to use ``appimagecraft`` over what I am doing with ``deb2appimage``?
This is what I am doing: https://github.com/guysoft/pi-imager/blob/qml/.github/workflows/build-linux.yml#L36

My ci/cd is currently building all 3 operating systems

TheAssassin, 2022-10-18:
Never heard of `deb2appimage`. appimagecraft itself is just a simple shell script generator. You should be asking "is there any advantage in using linuxdeploy". linuxdeploy and its plugins are by far more widely in use, and they try to avoid the hacks used by deb2appimage, pkg2appimage and alike.

lurch, 2022-10-18:
Sorry... :wink: 
https://xkcd.com/927/

tdewey-rpi, 2024-02-06:
Closing as won't fix. AppImage building was rejected in the PR, and as such we can't offer support for it.

If you choose to use a packaging scheme other than the ones provided by our releases, note that my policy is to refuse issues that appear to be caused by packaging misconfiguration (when you're not using our packages).

closed 2024-02-06 by tdewey-rpi

TheAssassin, 2024-03-01:
You may want to reopen this given the fact that #495 was reopened.