raspberrypi/rpi-imager #351: Can't get OpenSSL detected on windows build

guysoft, 2022-02-06:
Steps to reproduce:

1. Install QT with minw81_32
2. Install git bash and checkout this repo
3. install Msys2 (64bit default)
4. install cmake for windows from cmake.org

5. Open Msys2 MingMW and run the following to get the 32bit dlls:
```
pacman -S mingw-w64-i686-toolchain
```

6. Extract from the path ``C:\msys64\mingw32\bin`` : ``libssl-1_1.dll``, ``libcrypto-1_1.dll`` and ``openssl.exe`` and copy them to: ``C:\Qt\5.15.2\mingw81_32\``

7. create a build folder in rpi-imager source folder, cd in to it and run ``cmake ..``

The error is:
```
CMake Error at C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY) (found
  version "1.1.1i")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
-- Configuring incomplete, errors occurred!
  C:/Program Files/CMake/share/cmake-3.22/Modules/FindOpenSSL.cmake:574 (find_package_handle_standard_args)
See also "D:/a/pi-imager/pi-imager/build/CMakeFiles/CMakeOutput.log".
  CMakeLists.txt:115 (find_package)
```

I also get the same error in this github action:
https://github.com/guysoft/pi-imager/blob/windows-ci/.github/workflows/build-windows.yml

How do I get OpenSSL to be recognized?

maxnet, 2022-02-06:
Out of the top of my head:

- cmake can also be installed through Qt installer
- openssl 1.1 needs to be compiled and installed from source. Besides that the dlls need to end up in mingw bin directory, it also needs headers, import .lib, etc. installed in include and lib dir.

guysoft, 2022-02-07:
@maxnet 
* You get libs and headers too, they are in ``\lib`` and ``\include``.
I tried also adding ``DOPENSSL_ROOT_DIR``, ``DOPENSSL_CRYPTO_LIBRARY``, ``DOPENSSL_INCLUDE_DIR`` and ``D_OPENSSL_VERSION``:
```
cmake -DOPENSSL_ROOT_DIR="C:\msys2\msys64\mingw32" -DOPENSSL_CRYPTO_LIBRARY="C:\msys2\msys64\mingw32\lib\libssl.a" -DOPENSSL_INCLUDE_DIR=C:\msys2\msys64\mingw32\include -D_OPENSSL_VERSION=1.1.1m ..
```

Perhaps I need to set the paths to be windows or unix? It should find those.

guysoft, 2022-02-07:
Ok, one more clue - it looks like it is finding the settings, because if I only run ``cmake ..`` it says its finding OpenSSL version 1.1.1i, but if I do set it it find version 1.1.1m, which is the one provided by mingw-w64-i686-toolchain

maxnet, 2022-02-07:
- I did not use any OpenSSL version that came with a toolchain. But build from source
- I do not recommend using "cmake" from the command line on Windows, try the simple way first.

```
Building can be done manually using the command-line, using "cmake", "make", etc., but if you are not that familar with setting up a proper Windows build environment (setting paths, etc.), it is easiest to use the Qt creator GUI instead.
````

Is in the README for a reason. There are more paths you need to worry about than just OpenSSL if you are not using Qt Creator which comes with nice defaults.

guysoft, 2022-02-07:
* I understand and would have preferred to build openssl (no idea how to automate manual signing), but I also got stuck building openssl, there was some compile error command:
```
gcc  -I. -Iinclude -m32 -Wall -O3 -fomit-frame-pointer -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_CPUID_OBJ -DOPENSSL_BN_ASM_PART_WORDS -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DRC4_ASM -DMD5_ASM -DRMD160_ASM -DAESNI_ASM -DVPAES_ASM -DWHIRLPOOL_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"C:/Program Files (x86)/Common Files/SSL\"" -DENGINESDIR="\"/d/a/pi-imager/pi-imager/openssl/openssl-1.1.1m/dist/lib/engines-1_1\"" -DUNICODE -D_UNICODE -DWIN32_LEAN_AND_MEAN -D_MT -DOPENSSL_USE_APPLINK -DNDEBUG  -MMD -MF crypto/dso/dso_vms.d.tmp -MT crypto/dso/dso_vms.o -c -o crypto/dso/dso_vms.o crypto/dso/dso_vms.c
gcc  -I. -Iinclude -m32 -Wall -O3 -fomit-frame-pointer -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_CPUID_OBJ -DOPENSSL_BN_ASM_PART_WORDS -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DRC4_ASM -DMD5_ASM -DRMD160_ASM -DAESNI_ASM -DVPAES_ASM -DWHIRLPOOL_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"C:/Program Files (x86)/Common Files/SSL\"" -DENGINESDIR="\"/d/a/pi-imager/pi-imager/openssl/openssl-1.1.1m/dist/lib/engines-1_1\"" -DUNICODE -D_UNICODE -DWIN32_LEAN_AND_MEAN -D_MT -DOPENSSL_USE_APPLINK -DNDEBUG  -MMD -MF crypto/dso/dso_win32.d.tmp -MT crypto/dso/dso_win32.o -c -o crypto/dso/dso_win32.o crypto/dso/dso_win32.c
crypto/dso/dso_win32.c: In function 'win32_load':
crypto/dso/dso_win32.c:98:5: error: unknown type name 'HINSTANCE'
   98 |     HINSTANCE h = NULL, *p = NULL;
      |     ^~~~~~~~~
crypto/dso/dso_win32.c:98:19: warning: initialization of 'int' from 'void *' makes integer from pointer without a cast [-Wint-conversion]
   98 |     HINSTANCE h = NULL, *p = NULL;
      |                   ^~~~
crypto/dso/dso_win32.c:106:9: warning: implicit declaration of function 'LoadLibraryA' [-Wimplicit-function-declaration]
  106 |     h = LoadLibraryA(filename);
      |         ^~~~~~~~~~~~
crypto/dso/dso_win32.c:107:11: warning: comparison between pointer and integer
  107 |     if (h == NULL) {
      |           ^~
crypto/dso/dso_win32.c:129:11: warning: comparison between pointer and integer
  129 |     if (h != NULL)
      |           ^~
crypto/dso/dso_win32.c:130:9: warning: implicit declaration of function 'FreeLibrary' [-Wimplicit-function-declaration]
  130 |         FreeLibrary(h);
      |         ^~~~~~~~~~~
crypto/dso/dso_win32.c: In function 'win32_unload':
crypto/dso/dso_win32.c:136:5: error: unknown type name 'HINSTANCE'
  136 |     HINSTANCE *p;
      |     ^~~~~~~~~
crypto/dso/dso_win32.c: In function 'win32_bind_func':
crypto/dso/dso_win32.c:163:5: error: unknown type name 'HINSTANCE'
  163 |     HINSTANCE *ptr;
      |     ^~~~~~~~~
crypto/dso/dso_win32.c:166:9: error: unknown type name 'FARPROC'
  166 |         FARPROC f;
      |         ^~~~~~~
crypto/dso/dso_win32.c:182:13: warning: implicit declaration of function 'GetProcAddress' [-Wimplicit-function-declaration]
  182 |     sym.f = GetProcAddress(*ptr, symname);
      |             ^~~~~~~~~~~~~~
In file included from crypto/dso/dso_win32.c:490:
/usr/include/w32api/tlhelp32.h: At top level:
/usr/include/w32api/tlhelp32.h:15:3: error: unknown type name 'HANDLE'
   15 |   HANDLE WINAPI CreateToolhelp32Snapshot(DWORD dwFlags,DWORD th32ProcessID);
      |   ^~~~~~
/usr/include/w32api/tlhelp32.h:15:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'CreateToolhelp32Snapshot'
   15 |   HANDLE WINAPI CreateToolhelp32Snapshot(DWORD dwFlags,DWORD th32ProcessID);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/w32api/tlhelp32.h:26:5: error: unknown type name 'SIZE_T'
   26 |     SIZE_T dwSize;
      |     ^~~~~~
```

* I also have windows vm with qt creator installed and open to the rpi-imager project, but I am getting the exact same openssl error there too.

maxnet, 2022-02-07:
>I understand and would have preferred to build openssl (no idea how to automate manual signing)

I am personally not a big fan of building applications automated in the cloud either.
By signing it you are pretty much saying that you are vouching for the integrity of the build.
And I trust the integrity of my local build virtual machine -that is not used for anything else than compiling stuff- more than a virtual machine at some cloud provider where I have no idea on how many of there staff members have access too...

Anyway, to answer your question.

>but I also got stuck building openssl, there was some compile error command (perhaps it might help to show this?):

I am afraid I do not have time to provide detailed assistance on how to setup a Qt build environment with OpenSSL.
But the general idea is:

- Get some msys. Think I had the i686 one from: https://repo.msys2.org/distrib/i686/  but may not matter use if you compiler that came with Qt.
- Set path to where your compiler is. E.g. something along the lines of: export PATH=/c/qt/5.14.1/mingw73_32/bin:/c/qt/tools/mingw7320_32/bin:$PATH
- Install Perl some way (pacman -S perl ?)
- ./configure --prefix=$PWD/some-temporary-folder no-idea no-mdc2 no-rc5 shared mingw
- mingw32-make.exe
- mingw32-make.exe install
- Get files from some-temporary-folder, sign them, copy to the [bin,include,lib] folders under c:\qt\someversion\mingw_something

guysoft, 2022-02-07:
I managed manually to get it to build with those ssl libs by 
1. commenting out:
```
# find_package(OpenSSL REQUIRED)
```
2. Adding at the end:
```
set(OPENSSL_INCLUDE_DIR "C:/msys64/mingw32/include")
set(OPENSSL_LIBRARIES "C:/msys64/mingw32/lib/libssl.a" "C:/msys64/mingw32/lib/libcrypto.a")
```

I also had to remove signtool.exe (I have it at ``C:\Program Files (x86)\Windows Kits\10\bin\x64``):
```
    # Code signing
    # find_program(SIGNTOOL "signtool.exe" PATHS "c:/Program Files (x86)/Microsoft SDKs/ClickOnce/SignTool")
    # if (NOT SIGNTOOL)
    #     message(FATAL_ERROR "Unable to locate signtool.exe used for code signing")
    # endif()
```

Then it builds and runs

guysoft, 2022-02-07:
Just saw your other message: Thanks for giving me direction so far.
ATM I want some form of ci/cd for the fork, so I am going to hopefully make it build without any signature and have something people can play with. Afterwards I can look in to signing and other stuff assuming this matures to be a fork by its own right.

closed 2022-02-07 by guysoft