raspberrypi/rpi-imager #352: [Feature Request] - Country Code / Detect Country

japharl, 2022-02-06:
All,

If possible, it would be nice to default the WIFI country to  a reasonable default (possibly based upon time zone) and / or instead of a pull down menu of abreviations, see the full country name.

$0.02, otherwise the imager is awesome.

Zak

maxnet, 2022-02-07:
Not sure if time zone is precise enough. 5 ghz wifi has a lot of regional differences.

If computer running Imager is connected to wifi, may be able to ask what is being used right now.
E.g. Can try “iw reg get” on Linux.
On osx I think one can get what the access point broadcasts as country code with “ /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s”

Just don’t know what to use on Windows.
(The public documentation does mention an API call to get all supported regions from the wifi card, but not what is actually being used.
https://docs.microsoft.com/en-us/windows/win32/api/wlanapi/nf-wlanapi-wlanqueryinterface )

pelwell, 2022-02-07:
I'm reluctant for the imager to do anything automatically, in case it chooses the wrong region. The fact that you can make this setting once, as opposed to having to configure each new image, is enough of a win.

maxnet, 2022-02-07:
>I'm reluctant for the imager to do anything automatically, in case it chooses the wrong region. 

Not fully automatic, just prefill it as the default in the selection box the user still has to confirm.
Is it that much different from piwiz filling in "United Kingdom" as default for the country selection box, while I am not there either?

pelwell, 2022-02-07:
"United Kingdom" is not a WLAN region/country code, which is what we are discussing. I don't think the FCC care which timezone you have set your device to, but a setting which affects RF emissions is definitely of interest.

maxnet, 2022-02-07:
>"United Kingdom" is not a WLAN region/country code

piwiz does end up putting "country=GB" in wpa_supplicant.conf based on that box...

https://github.com/raspberrypi-ui/piwiz/blob/6a4455c51fe3244405cab1e12a7f8950bd21464e/src/piwiz.c#L1575-L1603

japharl, 2022-02-07:
Perhaps another api is available.  When you install windows, selecting your
country is the first step of the install process...  (a quick glance of the
win32 api didnt show much available under internationalism options.)

pelwell, 2022-02-07:
I think there's an important difference between the locale of the system running the updater and the device that is being configured - I'm thinking about a laptop that is roaming away from home - but @ghollingworth may have other ideas.

japharl, 2022-02-07:
I guess it depends on what reasonable defaults are.

tdewey-rpi, 2024-02-06:
Rejecting this feature request.

Not that I think it's a bad idea - I'd love to be able to offer this. Unfortunately, it requires reading the mind of the person using Imager regarding the end deployment of the Raspberry Pi to catch all the exotic edge cases - some of which might result in concrete real-world problems.

lurch, 2024-02-06:
> Unfortunately, it requires reading the mind of the person using Imager

I guess we'll need to wait for the mind-reading peripheral in Raspberry Pi 7 to be released first :brain: :rofl: