raspberrypi/rpi-imager #353: [Feature Request] Apple Silicon build

glefand, 2022-02-07:
Would it be possible to make an universal build of the rpi-imager supporting Apple Silicon (M1) ?

maxnet, 2022-02-07:
Problem is that an universal build will not work with older versions of Mac OS X.
And I am still not convinced the slight benefit of Imager also working if someone has objections against installing Rosetta outweigh that of excluding users with older OS versions, or the hassle with having multiple OSX Imager downloads (which may confuse some users).

There is no difference in the time it takes to write a SD card when Imager is run under Rosetta or native. Any increased CPU usage is not a limiting here.
And I don't think the arguments I heard so far of people reading somewhere on the Internet that it affects battery life or that running it under a profiler does show a performance difference, are very convincing.

glefand, 2022-02-07:
Agree with all your comments, except Rosetta will not last forever.
the sooner it can be done the better it is.
Having 2 builds (with the x89 version proposed by default) could be the easiest solution.

maxnet, 2022-02-07:
>Agree with all your comments, except Rosetta will not last forever.

Don't see the need to solve a problem now, that is not there yet.

The older Mac hardware will not last forever either, and there may come a point where the number of users with older OSX becomes so low that it makes more sense to stop providing builds compatible with that.

cperezabo, 2022-02-19:
In my case Raspberry Pi Imager is one of the 3 apps in my workflow with no M1 native builds. 
So @maxnet, for what I can understand, you are asking me and many other people to install Rosetta (a temporary solution to support **out of date** software) just because you don't want to simply add another pipeline to build the universal binary. That isn't more than a subjective whim, there is not a single valid technical reason not to do it.

maxnet, 2022-02-19:
>In my case Raspberry Pi Imager is one of the 3 apps in my workflow with no M1 native builds.

So you already needed Rosetta for your 2 other apps...

>So @maxnet, for what I can understand, you are asking me and many other people to install Rosetta (a temporary solution to 
>support out of date software) just because you don't want to simply add another pipeline to build the universal binary.

Problem is not compiling it, but providing support for the build, and dealing with users installing the wrong version if we offer multiple builds.

- Do not currently have M1 gear, so cannot properly test it. Although I have been told that the test build I compiled before, does work.
- An universal build would need Qt 6, which requires at least Mac OS X 10.14.
The market share of Mac OS X versions older than that is currently still around 10% (according to: https://gs.statcounter.com/macos-version-market-share/desktop/worldwide )
That is still quite a lot of people.
Even if you put a large "do not use this download on older Mac OS X versions, use the other one" message on the download page, some users are going to end up downloading the wrong version, and will complain to us about it not working...

vast-z, 2022-04-04:
Apple Silicon is already supported by a large number of software, and this is a trend of the future. It is also necessary to build a separate universal version, looking forward to the author's support

glefand, 2022-04-20:
"Problem is that an universal build will not work with older versions of Mac OS X.
And I am still not convinced the slight benefit of Imager also working if someone has objections against installing Rosetta outweigh that of excluding users with older OS versions, or the hassle with having multiple OSX Imager downloads (which may confuse some users)."

FYI, an app like VLC http://www.videolan.org/ offers multiple build, including Apple silicon build, without any issue or confusing users.

glefand, 2022-04-25:
FYI, even Microsoft is moving and providing a MSteams Universal binary. https://statics.teams.cdn.office.net/production-osx/1.5.00.11505/Teams_osx.pkg

now waiting for rpi-imager , at least, apple silicon build :-)

robd003, 2022-09-03:
@maxnet any update on making an arm64 build for macOS?

My office has replaced all Intel Mac's with ARM systems, I imagine lots of other places are too.

maxnet, 2022-09-03:
>@maxnet any update on making an arm64 build for macOS?

No plans for that.

cperezabo, 2022-09-04:
I feel embarrassed for the excuses given above. This software resulted to be nothing else but a toy. We are ending 2022, almost every project, even Windows is moving to arm and you try to justify that you cannot move this tiny app?

maxnet, 2022-09-04:
>I feel embarrassed by the excuses given above. This software resulted to be nothing else but a toy. We are ending 2022, almost 
>every project, even Windows is moving to arm and you try to justify that you cannot move this tiny app?

This project includes source code.
You are welcome to compile it for your favorite platform, and provide support to others on it.

cperezabo, 2022-09-05:
Yo sell boards, you give customers an official tool, and one day you decide you don't want to work so much so you leave your customers on their own. Sweet.

ghollingworth, 2022-09-06:
As Maxnet has already pointed out we do not support this architecture right now.

closed 2022-09-06 by ghollingworth