raspberrypi/rpi-imager #358: Updates to config.txt for embedded imager

peterharperuk, 2022-02-11:
Can we remove the following - DWC is deprecated evidentially.

dtoverlay=dwc2,dr_mode=host

Instead we should add so things work on CM4...

otg_mode=1

And it's worth also adding this...

arm_boost=1

maxnet, 2022-02-11:
>otg_mode=1

Is it no longer necessary to force host mode to deal with boards with broken otg?

timg236, 2022-02-11:
For backwards compatibility the DWC2 controller is still the default. However, in Raspberry Pi OS we have the following in `config.txt`

```
[cm4]
otg_mode=1

```

maxnet, 2022-02-11:
Ok, will change that.

My fault for not looking up what the option actually does.
(Wrongfully assumed an option called "otg_mode" would activate the USB controller in an OTG mode when set to 1.
But now understands it muxes the USB-C port to a XHCI host only controller instead, that does not do any OTG negotiation. Then there indeed shouldn't be any problem with carrier boards where otg is broken)

lurch, 2022-02-11:
IIRC forcing the OTG-mode is a Linux kernel option (so it'd be specified in `cmdline.txt`) rather than a firmware (`config.txt`) option?

maxnet, 2022-02-11:
>IIRC forcing the host/device OTG-mode is a Linux kernel option

With dwc2 that was done in config.txt as last part of the dtoverlay=dwc2 line.
With ",dr_mode=host" to force host mode
",dr_mode=peripheral" to force gadget mode.
If not specified it would default to ",dr_mode=otg" which means use OTG Host Negotiation Protocol to decide (which may not work if the right pin is not hooked up properly on the carrier board).

See also the dwc2 kernel device-tree docs.
https://www.kernel.org/doc/Documentation/devicetree/bindings/usb/dwc2.txt

>- dr_mode: shall be one of "host", "peripheral" and "otg"

So that is where my confusion came from.

lurch, 2022-02-12:
You're right, I was getting confused myself https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget
Apologies!

closed 2022-02-14 by maxnet

gabyavra, 2023-02-15:
how to configure cm4 to be used as a USB Gadget (ethernet via the same usb-c that is used for power)? thank you

lurch, 2023-02-15:
> how to configure cm4 to be used as a USB Gadget (ethernet via the same usb-c that is used for power)? thank you

That sounds like a question for the https://forums.raspberrypi.com/ , not for the RPi Imager repo.

commit 0cde6c2d (2022-02-14): Embedded build: use XHCI host-only USB controller block on CM4

Closes #358
commit 81b48574 (2022-02-14): Embedded build: use XHCI host-only USB controller block on CM4

Closes #358