raspberrypi/rpi-imager #359: rpi-imager v1.7.1 fails with permission denied when ran from i3 in Ubuntu

quatrox, 2022-02-11:
The snap packet for rpi-imager does not ask for sudo password when ran from i3. This causes a permission denied error.
It works when ran via GNOME3.

This took me a lot of effort to figure out.
I believe this is the same issue as https://forums.raspberrypi.com/viewtopic.php?t=273350.

maxnet, 2022-02-11:
Imager does NOT need root, on normal desktop systems that have udisks2.
Is that missing on i3?

lurch, 2022-02-12:
We've seen several issues with the snap build of RPi Imager (which is maintained by a 3rd-party) which seem to go away when you use the `.deb` version from https://www.raspberrypi.com/software/ instead?

nim65s, 2022-09-15:
Hi,
Getting the same issue on ArchLinux / Sway, while it works on XFCE here.

I think this is because i3 / sway don’t have a default polkit authentication agent: 
https://wiki.archlinux.org/title/Polkit#Authentication_agents

I get rpi-imager "write" button correctly working on sway just by manually starting one, eg. `/usr/lib/polkit-gnome/polkit-gnome-authentication-agent-1` from the `polkit-gnome` package.

tdewey-rpi, 2024-02-06:
Closing as 'Won't fix'.

Thanks, @nim65s for your sleuthing. This is not an issue that can be corrected in the rpi-imager codebase, and is a matter of configuring your desktop accordingly.

closed 2024-02-06 by tdewey-rpi