raspberrypi/rpi-imager #360: make fails on

gerrycreager, 2022-02-12:
When I tried to compile on Oracle Linux 8 (now that Centos is no longer LTS-reliable), I see this:

Updating '/home/gerry.creager/RPi/Imager/rpi-imager/rpi-imager_fr.qm'...
    Generated 75 translation(s) (72 finished and 3 unfinished)
    Ignored 66 untranslated source text(s)
[ 37%] Automatic RCC for translations.qrc
[ 40%] Building CXX object CMakeFiles/rpi-imager.dir/rpi-imager_autogen/mocs_compilation.cpp.o
In file included from /home/gerry.creager/miniconda3/x86_64-conda-linux-gnu/include/c++/9.3.0/chrono:41,
                 from /home/gerry.creager/miniconda3/include/qt/QtCore/qobject.h:59,
                 from /home/gerry.creager/miniconda3/include/qt/QtCore/QObject:1,
                 from /home/gerry.creager/RPi/Imager/rpi-imager/rpi-imager_autogen/EWIEGA46WW/../../cli.h:4,
                 from /home/gerry.creager/RPi/Imager/rpi-imager/rpi-imager_autogen/EWIEGA46WW/moc_cli.cpp:9,
                 from /home/gerry.creager/RPi/Imager/rpi-imager/rpi-imager_autogen/mocs_compilation.cpp:2:
/home/gerry.creager/miniconda3/x86_64-conda-linux-gnu/include/c++/9.3.0/ctime:80:11: error: '::timespec_get' has not been declared
   80 |   using ::timespec_get;
      |           ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/rpi-imager.dir/build.make:173: CMakeFiles/rpi-imager.dir/rpi-imager_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/rpi-imager.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

I also tried getting back to the default python with the same result.

maxnet, 2022-02-12:
>In file included from /home/gerry.creager/miniconda3/x86_64-conda-linux-gnu/include/c++/9.3.0/chrono:41,

That does not look like a clean installation of Oracle Linux 8
Please get rid of anything mentioning your home directory in your PATH and other environment variables, and try again.

gerrycreager, 2022-02-13:
Thanks, I'll try that. Thought I'd done so, but something wasn't happy.
Gerry

tdewey-rpi, 2024-02-06:
Closing as stale.

closed 2024-02-06 by tdewey-rpi