raspberrypi/rpi-imager #361: Feature request - gadget mode options

WilliamNoad, 2022-02-13:
I'd like to see a way to configure gadget modes with

- serial port
- Ethernet with optional static IP

The Pi Zero makes a useful tool for teaching shell scripting, Python, SQL, Perl, C/C++. Doing that through a basic CLI environment changes the teaching dynamic to focus much more on the subject matter than the application of some set of GUI tools.

maniflopi, 2024-05-20:
I support the interest to parameterize the static ip

tdewey-rpi, 2024-08-28:
Feature request rejected.

Raspberry Pi *Imager* isn't the right tool for this, but a custom image built with *pi-gen* or *pi-gen-micro* absolutely is.

Expect more documentation in this area in the future.

closed 2024-08-28 by tdewey-rpi