raspberrypi/rpi-imager #368: Unable to boot raspberry pi 3 B+ after power off before shutdown

saad-addverb, 2022-02-17:
I am installing rasbian using rasberry pi imager. but first time it is getting started and I did the initial settings like language, wifi etc. after that I rebooted it and getting error as follows:

[end kernel panic - not syncing: VFS: unable to mount root fs on unknown -block(179,2) ].

after that I erased the data of sd card and again installed the rasbian from rasberrypi imager, this time i am unable to use sudo command and getting the error as follows whenever i am running any command:

syntax error /etc/apt/apt.conf.d/70debconf:1: Extra junk at end of the file

lurch, 2022-02-17:
Sounds like perhaps your SD card is failing and giving you random errors? Safest option is to probably just buy a new SD card.

tdewey-rpi, 2024-02-06:
Closing as won't fix. This isn't an issue with rpi-imager.

closed 2024-02-06 by tdewey-rpi