raspberrypi/rpi-imager #369: Error creating firstrun.sh on FAT partition on Ubuntu

EddieTheHead, 2022-02-17:
Write process ends with "Error creating firstrun.sh on FAT partition".
Version: v1.7.1
Host os: Ubuntu 20.04.3 LTS
RPi os: RASPBERRY Pi OS (32 bit)
Options:
Enabled ssh, changed locale to pl/Warsaw, added custom password, added WiFi credentials. 
[image]
[rpi.log](https://github.com/raspberrypi/rpi-imager/files/8088851/rpi.log)

maxnet, 2022-02-17:
>/snap/rpi-imager/221/usr/local/bin/rpi-imager

- Uninstall snap.
- Get the .deb from the releases page.

closed 2022-02-17 by maxnet

CatoAntonsen, 2022-03-04:
Why is sthis closed? I have the same error, same version on Ubuntu  (20.04.4 LTS)

maxnet, 2022-03-04:
>Why is sthis closed?

Because the log file of the user shows he is not using Raspberry Imager, but Canonical's Imager variant which is not provided by us.

>I have the same error, same version on Ubuntu (20.04.4 LTS)

And your Imager did came from a .deb file that you downloaded from the raspberrypi.com site or from the "releases" page here?

CatoAntonsen, 2022-03-04:
I download it via raspberrypi.com > Software > Imager > Download for Ubuntu a86). When downloaded, FireFox opens it with Software Install (default). Failed to install file: Not supported.

 Ehere is "releases?

maxnet, 2022-03-04:
> FireFox opens it with Software Install (default). Failed to install file: Not supported.

Not sure what application "Software install" is exactly.
If it's Ubuntu's ownsoftware manager, or something else accidently got associated with .deb files on your system.

Do you have a screenshot of it, complete with error message?

Alternatively, save the .deb to file first, and run: "sudo apt install ./rpi-imager_1.7.1_amd64.deb" from terminal.

>Ehere is "releases?

rpi-imager_1.7.1_amd64.deb at https://github.com/raspberrypi/rpi-imager/releases

CatoAntonsen, 2022-03-04:
Thank you guys! It's working now! :-)

simonschllng, 2022-04-01:
> but Canonical's Imager variant which is not provided by us.

Why are there always so many unofficial variants which just cause pain?

Thanks all for clarification. Annoying problem.