raspberrypi/rpi-imager #372: FEATURE REQUEST:  Add support for Windows 10/11 Compressed Folder Compression

Jibun-no-Kage, 2022-02-20:
FEATURE REQUEST:  Add support for Windows 10/11 Compressed Folder Compression.  The current version of rpi-imager 1.7.1 is great, but it can't read image files that are compressed as Windows 10/11 native zip files.  Current version of utility reports that zip files are in an unknown or unsupported compression, "Error extracting archive.  Unsupported ZIP compression method (9: deflation-64-bit)."

lurch, 2022-02-20:
There's some useful (maybe?) info about "compression method 9" in https://github.com/thejoshwolfe/yauzl/issues/58

I guess in the meantime you'll just have to uncompress your image first before loading it into RPi Imager.

maxnet, 2022-02-20:
Don't think we should support proprietary .zip flavors.
Politics aside (zlib and libarchive refusing to support it), if we did add it, we would have the practical problem that images may get into distribution that could only be written by recent Imager versions, and will fail for others.

Jibun-no-Kage, 2022-02-20:
The goal of this request is such that your utility can be more user friendly and flexible on Windows 10/11.  A core feature of Windows is compressed files but you have a Windows application that can't support such.  A windows user is presented with a confusing and limiting issue, in a Windows application.  IMHO, if you going to create an application for an given OS, support the core features that OS presents.  The majority of Windows users are not going to care about what compression formats are compatible with Linux, just presented with a confusing limitation, the perception being that your Windows application is inflexible.  The goal of the entire Pi device effort was education no?  And yet you create a tool that is limited and confusing to new users?  Should this not be avoided?

maxnet, 2022-02-20:
Eh, wouldn't NEW USERS be more likely to use a ready-made image from our repository, instead of creating their own images?

Jibun-no-Kage, 2022-02-20:
Good question, and point.  Very new users would yes.  But Windows users don't think outside of Windows... in fact Windows users are encouraged to stay in the bubble of Windows, and compressed folders has a long history and tradition.  As some one that has be multi OS (CMP, DOS, HP Unix, Unix, Mainframe variants, PDP-11, VAX, Red Hat, CentOS, Debian 6.x forward, Amiga, Apple, MacOS, Windows 1x forward,, NeXT OS, etc., etc.), since I was 17... and I am now over 57... (cough) I am still surprised how Macintosh users and well as Windows users tend to stay in their respective bubbles exclusively and consistently.

In reference to adding image capture to the current tool, now that Windows supports native) tar/tar.gz file creation, I suggest again that you consider adding image capture, so the tool is able to create images as well was use existing (compressed etc.).  Being able to create images, which would use native compression and archiving, via Windows native tar/tar.gz support, would avoid the entire need to support deflate64 directly.  One stop shop if you will.  Unfortunately, this still does not provide the painless transparent support of explorer compressed files, as I hope would be supported.  But it would promote using tar.gz format as a dominate de-facto standard, which would be consistent across multiple platforms/OSes.

Jibun-no-Kage, 2022-02-20:
Oh, and not to sit on my thumbs, but cranked out the following, if a bit dirty batch/cmd file, to generate compressed image file that the imager will accept.  Not using Explorer native compressed folder method, but one fully supported under Windows 10/11.

`@ECHO OFF
@CLS

::

@SET _Out_=%~1%.tar.gz
@SET _Log_=%_Out_%.log
@SET _Sum_=%_Out_%.sum
@SET _In_=%~1%

@SET _Title_=MakeTarGZip
@SET _Version_=1.0.1
@SET _Release_=02/19/2022

::

@IF EXIST "%_Out_%" DEL "%_Out_%"
	@IF EXIST "%_Sum_%" DEL "%_Sum_%"
		@IF EXIST "%_Log_%" DEL "%_Log_%"

@ECHO %_Title_% * %_Version_% * %_Release_%

::

:_Compress

	::

	@ECHO Tar And GZip '%_In_%'...

		@TAR -zcvf "%_Out_%" "%_In_%" >> "%_Out_%.log" 2>&1
		@SET _Result_=%ERRORLEVEL%

	@ECHO ...Tar And GZip '%_Out_%' (Result: %_Result_%).

		@IF NOT %_Result_%==0 GOTO _Error

:_Hash

	::

	@ECHO Hash '%_Out_%'...

		@CERTUTIL -hashfile "%_Out_%" sha256 >> "%_Out_%.sum" 2>&1
		@SET _Result_=%ERRORLEVEL%

	@ECHO ...Hash '%_Out_%' (Result: %_Result_%).

		@IF NOT %_Result_%==0 GOTO _Error

:_Continue

	@DEL "%_Out_%.log"
@REM	@DEL "%_In_%"
	@GOTO _Exit

::

:_Error

	@IF EXIST "%_Out_%.sum" DEL "%_Out_%.sum"
	@PAUSE

:_Exit

	@EXIT`

lurch, 2022-02-21:
Interestingly, it seems sometimes "Compressed Folders" uses the regular DEFLATE, and other times it uses the less-widely-supported DEFLATE64 https://superuser.com/questions/1249267/how-does-windows-decide-whether-to-use-deflate-or-deflate64-for-zipping-files

Looks like RPi Imager isn't the only Windows-program that this causes problems for... https://www.google.com/search?q=windows+compressed+folders+DEFLATE64

Jibun-no-Kage, 2022-02-21:
As I recall anything under 2GB deflate is used and above that deflate64 is used?  If memory serves.  Ah, yes, the other link references the same as well.

lurch, 2022-02-21:
> all kinds of interesting, sometimes crazy things Microsoft code does at times

I spotted this interesting article yesterday https://devblogs.microsoft.com/oldnewthing/20180515-00/?p=98755
"The compression and decompression code for Zip folders was licensed from a third party. This happened during the development of Windows XP. This means that the feature set of Zip folders was locked to whatever features were hip and cool as of around the year 2000."

Jibun-no-Kage, 2022-02-21:
In a word... NUTS.

tdewey-rpi, 2024-02-06:
Closing as 'Won't fix'.

I fully expect OS vendors to be providing images in compressed formats they have tested with rpi-imager - and supporting exotic Microsoft-favoured formats is not currently planned.

closed 2024-02-06 by tdewey-rpi