raspberrypi/rpi-imager #375: [OS request] Add ioBroker Raspberry OS Image to the rpi-imager OS list

buanet, 2022-02-22:
Hi, I'm the maintainer of the official [ioBroker Raspberry OS image](https://github.com/buanet/ioBroker.raspberry-os) as well as the official [ioBroker Docker image](https://github.com/buanet/ioBroker.docker).

As noted [here](https://www.raspberrypi.com/news/raspberry-pi-imager-imaging-utility/#comment-1523756), I open this issue to request you to add our ioBroker Raspberry OS Image to the rpi-imager OS list.

IoBroker is a open source IoT platform written in JavaScript that easily connects smarthome components from different manufactures. With the help of plugins (called: "adapters") ioBroker is able to communicate with a big variety of IoT hardware and services using different protocols and APIs. All data is stored in a central database that all adapters can access. With this it is very easy to build up logical connections, automation scripts and beautiful visualizations. For further details please check out [iobroker.net](https://www.iobroker.net).

In order to make all necessary information available I already added a [rpi-imager.json](https://github.com/buanet/ioBroker.raspberry-os/blob/main/rpi-imager/rpi-imager.json) to the ci process so it will always provide the data of the latest release. 

I checked if it is working by starting the rpi-imager with `--repo https://raw.githubusercontent.com/buanet/ioBroker.raspberry-os/master/rpi-imager/rpi-imager.json`. Looks good so far. :)

[image]

So it would be very nice if you can add our ioBroker Raspberry Pi image at the `Other specific-purpose OS` > `Home assistants and home automation` section with this small snippet:

```
{
  "name": "ioBroker",
  "description": "Official Raspberry Pi OS Image for ioBroker open-source IoT platform.",
  "icon": "https://raw.githubusercontent.com/buanet/ioBroker.raspberry-os/master/rpi-imager/iob-icon.png",
  "subitems_url": "https://raw.githubusercontent.com/buanet/ioBroker.raspberry-os/master/rpi-imager/rpi-imager.json"
},
```

Thank you!

Regards,
André

jjm001, 2022-02-22:
Great idea!
👍

Horgul123, 2022-02-22:
Would be great!

ghollingworth, 2022-02-22:
Is there a reason why this isn't just a .deb file you can install?

simatec, 2022-02-22:
Great! 👍🏻

TepMan, 2022-02-22:
Great idea 👍

JogiBarbarski, 2022-02-22:
Sehr gute Idee. Habe da schon lange drauf gewartet.👍🏻👍🏻👍🏻

buanet, 2022-02-22:
@ghollingworth Of course, we also provide a setup script to install ioBroker under linux in general. But a ready to go image is a highly requested feature from the community. It's the best option for new (inexperienced) users to easily test ioBroker during their way into home automation.
As more and more people diving into smarthome topic we think it is important to make simple solutions available for everyone. There are a lot of people starting their Raspberry Pi experience when setting up some smarthome equipment. :)

Another thing is that the image gives us the possibility to configure what fits best for a dedicated ioBroker host. For example, we use the lite image (so no desktop environment by default) and configure things like boot target and memory split to increase performance. 

Regards,
André

FoGBaV, 2022-02-22:
Top idea!

seren2, 2022-02-22:
Sehr gute Idee..👍🏻

SteveEmperor, 2022-02-23:
Sehr gut
👍

Apollon77, 2022-02-23:
@ghollingworth Additionally in my eyes the update procedures for a nodejs based application do not really match to the "standard" people expect when installing a deb package. SO the package would only be there or an initial install, because after this anything is handled by npm as package manager. Especially because of the very modular approach of ioBroker and all plugins ( that can all be updated independently. Because of this it would be a very strange experience in my eyes that do not match to "how deb packages works". Better not mix it

joergzdarsky, 2022-02-23:
Very good idea, I support this!

wendy2702, 2022-02-23:
Good idea!

ghollingworth, 2022-02-24:
Also adding "Good idea" won't change anything!  What's really really bad is if I find out someone communicated with their users encouraging them to do this.  That's tantamount to spamming me which will earn you a ban

Apollon77, 2022-02-24:
@ghollingworth No one communicated to users to "spam" this post. Noone ever would do that. We just asked users to like the main post IF they find that idea to add ioBroker here is making sense for them for an easier future. And I think this is allowed, or?

And the two direct responses to your question above should also not being considered as spam I hope.

buanet, 2022-02-24:
@ghollingworth @XECDesign I'm totally with you. Agree/ disagree comments will not help in any way. The preferred way should be reacting with an emoji or maybe join the discussion with some arguments pro or con the os request... 

You can make sure that nobody called for spamming. As our devs working on ioBroker core and 483 plugins (aka adapters) we definitely know what it means to get spammed with feature requests or "when will it be available?"-posts.
That's one reason why we try keeping other devs and the community always up to date when working on requested features.

As you can see the most users know how it is working. So far there are 225 "supporters". Only 9 did not find the emoji button.... Does not really sounds like the big spam wave for me . ;) I hope you can also see it this way....

Regards,
André

Apollon77, 2022-12-17:
@ghollingworth @XECDesign Is there any chance to get added? In the meantime many other OS Requests were added, most of them without a single "like" - including Openhab as another smart home software.

Is there anything we need to do to have a chance to get added or should we not expect support? If not please state it with the reasons so that we know it and can react accordingly

Thank you for a statement.

Ingo

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.