raspberrypi/rpi-imager #377: v1.7.1 not ejecting boot partition if using advanced options

jpsingleton, 2022-02-23:
v1.7.1 on Ubuntu Linux

If eject media when finished is selected but customization options are set then the boot partition remains mounted at the end after the completion message. As this is only the boot partition, I suspect it is being mounted to make configuration changes at the end and then not being unmounted again.

lurch, 2022-02-24:
Are you using the snap version, or have you installed the deb version from https://www.raspberrypi.com/software/ ?

maxnet, 2022-02-24:
Ejecting is not implemented on Linux regardless of version.

jpsingleton, 2022-02-24:
OK thanks, I'll just always unmount manually before removing the SD card. Maybe change the success message on Linux? I'm using the official `.deb`.

I wasn't aware of the snap but will check it out as that will make updating easier. I assume it is this one: https://github.com/popey/imager-snap

I guess it might also be possible to add the Raspberry Pi OS apt repo to Ubuntu and install/upgrade it from there?

closed 2022-02-24 by jpsingleton

lurch, 2022-02-24:
The reason I asked about snap / deb is because people using the snap version often have errors, which disappear when they uninstall the snap version and install the deb version instead.

> I guess it might also be possible to add the Raspberry Pi OS apt repo to Ubuntu and install/upgrade it from there?

I _suspect_ the Raspberry Pi OS apt repo only contains arm and arm64 binaries? And I think the snap version is only x64?

jpsingleton, 2022-02-24:
> I suspect the Raspberry Pi OS apt repo only contains arm and arm64 binaries?

I'd guess so if it's not shared with Raspberry Pi Desktop for PC and Mac too.

lurch, 2022-02-24:
ping @XECDesign - is RPi Imager included in the repos for "Raspberry Pi Desktop for PC and Mac"?

XECDesign, 2022-02-24:
> ping @XECDesign - is RPi Imager included in the repos for "Raspberry Pi Desktop for PC and Mac"?

Yup

XECDesign, 2022-02-24:
Actually there should be an asterisk there. The Buster version would be old and there's no Bullseye release of Raspberry Pi OS for PC yet. The packages should be in the bullseye repo though.