raspberrypi/rpi-imager #378: OS Request FluiddPI

DarkMagicSource, 2022-02-25:
Similar to Mainsail and the varients of OctoPrint, [Fluidd](https://docs.fluidd.xyz/) is a different 3D Printer control software.

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi

jedahan, 2023-11-17:
I was going to submit this to the form, but was wondering if someone already submitted it - don't want to submit duplicates, so it would be nice to know what has been requested, and what hasn't.

tdewey-rpi, 2023-11-17:
We'll filter out duplicates or existing items as we process the responses.

jedahan, 2023-11-17:
Full transparency - I’d rather not go through the work of submitting if someone else already has and it’s in your queue. Is there a way to make the form responses read-only visible? Heck even just a column for the name of the submitted distros.

jedahan, 2023-11-17:
(If this is intentional as a filter for “is there enough interest for us to spend the work to maintain this district list”, ignore my request)

tdewey-rpi, 2023-11-17:
Unfortunately I can't open up the results, as part of the form asks for personally identifiable information, which is collected to help coordinate future Imager changes.

That's also why the form is now a required part of adding new items to the OS list. The device filtering changes are still incomplete because we didn't have a means to contact individuals capable of making the corresponding 2P changes.

In short, without a submission we won't be able to add your OS to the list.

jedahan, 2023-11-17:
Appreciate your patience explaining things.

lurch, 2023-12-20:
@jedahan And just to clarify, I believe the form is intended to be used by the developers / maintainers of a particular OS image, not by 3rd-party users requesting that their favourite OS gets added to RPi Imager :wink:

jedahan, 2023-12-20:
and fluiddpi is archived, so no need in this case unless someone revives it