raspberrypi/rpi-imager #380: Please Add multi SD card write

MkLHX, 2022-03-01:
Hello, thank you for the latest release with the parameters section it's soo helpful!

Is it possible to add the feature to select multiple SD card target for one write?

I mean, I need to create many SD cards with OS and minimal configuration (like the advanced options allow us now).

ghollingworth, 2022-03-02:
What is it that you are trying to do here?

Why do you need to program more than one card at a time?

This isn't really suitable for mass provisioning, since things like passwords will all be the same.

MkLHX, 2022-03-02:
The goal is to massively burn sd cards like balena etcher allowing us but with the Advanced Options like RPI Imager.
I need to burn sd cards with settings and config script to make many devices using RPI as heart.

maxnet, 2022-03-02:
>to make many devices using RPI as heart.

It wouldn't be possible to let those devices write their own SD card?
https://github.com/raspberrypi/cmprovision supports customization through post-install scripts.

MkLHX, 2022-03-02:
Thanks for this link but actually the RPI we used is RPI 3A+

tdewey-rpi, 2024-02-06:
Closing as 'Won't fix'.

As Gordon identified, this mechanism is absolutely not what you're looking for. If you're looking to mass provision devices, you need to investigate creating a custom image using pi-gen or similar. Using rpi-imager for this may result in shared passwords and authentication secrets - which pose a security risk to the entire family of devices.

closed 2024-02-06 by tdewey-rpi