raspberrypi/rpi-imager #383: RPi Imager writing to EmmC on CM4 - config.txt does not exist

WanaGo, 2022-03-07:
Hello,

Basically the same issue that has been written about here: https://github.com/raspberrypi/rpi-imager/issues/289
However using CM4 and the built in EMMC (CM4102016)

Writes to the emmc, gets to the end and verifies, and then says it cant write to the config.txt file, something about the file not existing etc. Drive is not present in Windows Explorer afterwards either.
I'm doing it again now to see if I get a different result. The drive was still not visible before I started, but the Imager managed to find the EmmC ok, and its writing again now.

Using Raspberry Pi Imager v1.7.1

Any ideas?

WanaGo, 2022-03-07:
[image]
Same again

maxnet, 2022-03-07:
>However using CM4 and the built in EMMC (CM4102016)

Does it behave different if you use a normal USB SD card reader or USB stick?

WanaGo, 2022-03-07:
> > However using CM4 and the built in EMMC (CM4102016)
> 
> Does it behave different if you use a normal USB SD card reader or USB stick?

Using a USB Drive, the process completes no problem.
[image]

Same when writing to a uSD card. 
Both systems boot with their appropriate bootloaders loaded.
However SSH does not work when you plug in the Ethernet, still have to put the blank ssh file on the /boot for SSH to work.
Wifi is not configured, neither is country locale etc.

Seems the settings applied in the Imager are just not making it to the card or emmc at all...

No SSH warning on the screen when using HDMI monitor, and Wifi is off, and the intro setup thing comes up despite the tick box saying to not run it.

WanaGo, 2022-03-07:
Pi4 beta bootloader, keyboard and blank uSD inserted, installed the 64-bit OS using the Pi itself, and set the same options for SSH and WiFi etc, and that worked totally fine, booted up and everything was set correctly.
But when I do it from 1.7.1 from the PC on the 32bit OS (ill try 64bit now) they just dont do anything.

Ill go back and try the CM4 also. Switched to Pi4 as its easier to use the card then emmc, but had the same issues.

WanaGo, 2022-03-08:
Yes just worked for me for the first time on the CM4 just now...
So weird, nothing has changed...

[image]

But it can clearly be seen the fail earlier with the CM4.

Oh well, its working now... just wish I knew why.

WanaGo, 2022-03-08:
Side note, is there any plan to integrate the rpiboot into imager, so you can just click a button or something inside imager when planning to write to a CM4 etc and it links it up, without having to run the rpiboot and then imager, etc?

lurch, 2022-03-08:
> Side note, is there any plan to integrate the rpiboot into imager, so you can just click a button or something inside imager when planning to write to a CM4 etc and it links it up, without having to run the rpiboot and then imager, etc?

IIRC Etcher supports that for CM1 and CM3, but I dunno if they support that for CM4 as well yet? :shrug:

maxnet, 2022-03-08:
>Side note, is there any plan to integrate the rpiboot into imager

The other way around do is possible.
One can rpiboot Imager to run on the CM4.

Provided you do have Ethernet/HDMI/mouse connected to the CMIO board

lurch, 2022-03-08:
> One can rpiboot Imager to run on the CM4.

Hmmm, interesting. Perhaps it's worth adding some step-by-step instructions to https://github.com/raspberrypi/usbboot/blob/master/Readme.md ? (ping also @timg236 )

maxnet, 2022-03-08:
Not sure if there is much demand for that.
There were instructions for it on the forum some while ago, in regards to using it to provision NVMe drives.

Anyway, just:
- obtain a build of Imager Embedded [somewhere](https://www.maxnet.eu/tmp/171-miscfixes/imager-embedded-20220223.zip)
- toss the files in c:\program files (x86)\raspberry pi\imager
- open a command prompt:

```
cd \program files (x86)\raspberry pi\imager
rpiboot -l -d imager
```

timg236, 2022-03-08:
The rpiboot-msd is going to be deprecated in favour of a Linux init-rd that opens exports all the block devices as USD MSD (using gadgetfs)

WanaGo, 2022-03-10:
Bah - just got it again on the CM4 writing latest 32bit OS using 1.7.1

[image]

maxnet, 2022-03-10:
If -after you get this message- you press [Windows key]+R, and start "diskpart"
Does "list volume" show H: belongs to some other partition than the FAT boot partition?

If no drive letter is assigned, and you do "list disk", note which disk number is your SD card, and then do:

```
select disk [number of your SD card]
select partition 1
assign
```

Does it successfully assign a drive letter, or do you get some error?

WanaGo, 2022-03-10:
diskpart will not even load
"DiskPart encountered an error starting the disk management services."

[image]

The H: is there but its not actually mounted when looking in explorer

[image]

Even in Computer Management, it will not load the virtual services - so something has messed up during this process, as this all worked fine earlier today.

I do notice at the end of the write/verify, it waits for the drive to mount, but it doesnt happen, when then leads to the error.

[image]

ill restart PC and see if I can get it to mount.
I did also try doing this process on my laptop and had the same sort of issues.

WanaGo, 2022-03-10:
[image]

This is the CM4, CM4102016, so 16GB emmc, no SD card.

So the emmc was written to, but has no partitions... and is not mountable...

WanaGo, 2022-03-10:
Got a fresh CM4 module, this one is a CM4101008 with 8GB of emmc, and this one worked out of the box like the 16GB one did, prompted to be formatted - so did that and then looked at diskpart and it all checks out fine, and H: was present.

[image]

Writing to this one now to see if it completes

maxnet, 2022-03-10:
>I do notice at the end of the write/verify, it waits for the drive to mount, but it doesnt happen, when then leads to the error.

It waits up to 3 seconds to see if Windows assigns a drive letter automatically.
If it did not do so within that time, it ask diskpart to assign the letter. (if automount is disabled, it does is normal Windows does not assign letter automatically, so the 3 second wait can be pretty normal).
That does do "something". As it apparently does give you a H:, that you probably did not had in the 3 seconds before.

Why H: is not functional, and why disk management service is not responding I do not know.
And don't think that is something we can easily debug. Being a closed source operating system and such.

WanaGo, 2022-03-10:
Same issue with the 8GB one

[image]

Just did the 16GB one on my laptop and same issue on that too

WanaGo, 2022-03-11:
It seems to write the emmc OK, I booted the CM4 after the failure of the config.txt and it booted fine. I then connected it to the PC again and H: showed up, and I was then able to do the mods like adding ssh. 
So I have no idea what is going on, why it fails consistently on both of my computers, doing the exact same thing.

lurch, 2022-03-11:
> So I have no idea what is going on, why it fails consistently on both of my computers, doing the exact same thing.

You mentioned that it worked correctly at least a couple of times, so I guess it's not 100% consistent? :shrug: 
Do you have some kind of add-on "security software" or AntiVirus that's common between both your computers?

WanaGo, 2022-03-11:
Yeah it worked correctly a couple of times.
I do have Comodo firewall on my desktop, and NOD32 Antivirus. But I had already thought of this and excluded RPi Imager from those, as I wondered if it was those stopping it too - but doesnt seem to be.
On my laptop I just just NOD32. So that is common I guess.
I can try disabling while doing this process and see if that works.

Yardkyn, 2023-03-29:
Guys, my problem was, that after first burning CM4, by imager it never been seen as storage. I wanted to burn another CM4, and the same problem for me solution was to do it by CMD
```
cd \program files (x86)\raspberry pi
rpiboot -l -d imager
```
and then, 
`rpiboot`
replug, format, and all done

jenkinz89, 2023-11-24:
> Guys, my problem was, that after first burning CM4, by imager it never been seen as storage. I wanted to burn another CM4, and the same problem for me solution was to do it by CMD
> 
> ```
> cd \program files (x86)\raspberry pi
> rpiboot -l -d imager
> ```
> 
> and then, `rpiboot` replug, format, and all done

I've had the same issue with CM3 and CM4. Initially I found that launching balenaEtcher in the background would then allow the module to be seen as storage (don't want to use BE as get lots of issues with failed flashes). 

Your work around seems to work fine for me.

tdewey-rpi, 2024-02-06:
Closing as stale - last substantial comment was March '22.

Please retest against the latest version before raising a new issue.

closed 2024-02-06 by tdewey-rpi