raspberrypi/rpi-imager #384: Pi4 set to SSH and WiFi on Imager 1.7.1, doesn't actually work

WanaGo, 2022-03-07:
Burning the latest OS to uSD for the Pi4, and setting up SSH and WiFi etc in Raspbery Pi Imager 1.7.1, does not seem to work when the card is put into the Pi.

SSH is not enabled.
WiFi is not enabled.
Once I got SSH working by taking the card out again and putting a blank ssh file onto the /boot and putting it back in the Pi again, and then I SSH into the Pi, the terminal looks like this:
[image]

Here are the Pi Imager settings:
[image]

Am I doing something wrong, or is this simply not working right?

First time using this feature, thought it would nice to have SSH enabled and set up WiFi before I get in so it works right off the bat, but its not working at all.

WanaGo, 2022-03-07:
> It did provided you with a "cog icon" button to do so? (As you can also access the advanced settings with a key combination, but settings will not be applied if using an OS image that does not support customization).

Cog icon yes. Latest 32bit OS, didnt use key-combinations, this is via PC so used keyboard and mouse v1.7.1 imager for windows.
 
> After pressing "OK" in the advanced settings, the advanced setting window did close? (If not, there may be a setting highlighted in RED that you incorrectly set)

Yes it closed. I also saved the settings and can go back in and all is set as per my picture above. No issues there.

> If after writing the customized image, you reinsert the card in your SD card reader again instead of sticking it in your Pi, did it create a firstrun.sh file on the FAT partition of your SD card?

Yes, image below.
 
> Concerns the same Windows computer as in your different issue about CM4?

Yes same computer. New a few months ago, Latest up-to-date Windows 10 Pro x64 21H1, Ryzen 9 5950X AMD processor, 64GB RAM, 3080 GPU, etc.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

This time it worked - SSH enabled, Wifi working, but I had just updated the bootloader to the beta... would that have made a difference?
Nothing changed in my settings, same uSD card, same Pi4, same computer, same everything other than new bootloader.

maxnet, 2022-03-07:
>This time it worked - SSH enabled, Wifi working, but I had just updated the bootloader to the beta... would that have made a 
>difference?

Negative.
firstrun.sh logic does not depend on the bootloader. Entirely handled by Linux/systemd.

If there was something very wrong with the SD card reader (driver) in your computer. E.g. if it did not finish writing the file to storage yet when we tell it to eject the card, and does not correctly flush the data to storage before doing so, than such things do could happen.
But that does not explain why it did not work correctly with CM4 either earlier.

d3-worgan, 2023-09-24:
Im having the exact same problem. I have tried editing the config in /netplan as described in ubuntu [guide](https://ubuntu.com/tutorials/how-to-install-ubuntu-on-your-raspberry-pi#4-boot-ubuntu-server) but no joy. Very odd.

d3-worgan, 2023-09-24:
Yes the imager doesnt actually work. Or maybe my Pi is broken? I have a Pi 3 and used the imager with Ubuntu Server 22.04 LTS 64 bit. The website says Raspberry pi 3 is officially supported. For some reason the wlan0 is not listed in /sys/class/net. I tried manually running the driver. In the boot logs I get an error saying the chip is not supported. Anyone have any suggestions?
[image]

maxnet, 2023-09-24:
>Ubuntu Server 22.04 LTS 64 bit

Suggest you try a different Linux distribution like RPI OS first before anything else...

closed 2024-02-06 by tdewey-rpi