raspberrypi/rpi-imager #385: [OS Request] DAKboard-OS

cocarrig, 2022-03-07:
Hi Team,  Thank you for taking the time to review this request.  We are hoping to add DAKboard-OS to the list of digital signage options made available in the Raspberry Pi Imager.  DAKboard is an easily customized display for calendars, weather, to-do lists and much more.  We offer free and paid services to our customers with a wide range of available integrations and recommend the Raspberry Pi for display devices with our OS available for free.

We would like to start by offering the 2.28 release which is available here for download and install currently:
[https://dakboard.freshdesk.com/support/solutions/articles/35000125880-download-and-install-the-dakboard-os-for-raspberry-pi](https://dakboard.freshdesk.com/support/solutions/articles/35000125880-download-and-install-the-dakboard-os-for-raspberry-pi)

Our release notes are located here:
https://github.com/dakboard/Hardware-OS/releases

Please reach out to contact@dakboard.com or ccarrigan@dakboard.com directly with any questions you have.

cocarrig, 2022-06-15:
Hi Team,

Would like to announce that the latest release available is now 3.0, which is available at the same link and provides a further simplified setup process.  Can we get an ETA for a review of this request to add DAKboard OS to the rpi-imager?

cocarrig, 2023-03-29:
Hi @ghollingworth and Team,

Are there other details needed to move this forward for review and further discussion?  Please feel free to contact me directly at ccarrigan@dakboard.com or indicate here.  I have seen other digital signage images added since I last looked and we would really love to add DAKboard to the lineup.

ghollingworth, 2023-03-29:
You'd need to provide a .json file which links to your images...

This is what we provide:

https://downloads.raspberrypi.com/os_list_imagingutility_v3.json

So you'd have to provide an entry similar to one of the other digital signage companies:, Infobeamer as an example

https://cdn.infobeamer.com/repo.json

Gordon

cocarrig, 2023-03-29:
@ghollingworth,

Perfect, I will get started on setting this up and reach back out to you once it's completed.  Thanks for the help!

ghollingworth, 2023-04-12:
> Hi @ghollingworth <https://github.com/ghollingworth>,
>
> If I were to provide you with a URL which would always contain the latest
> json data for our most current release, will the RPI-Imager automatically
> update? Thanks

Yes that's the idea of the .json system

cocarrig, 2023-05-02:
Hi @ghollingworth 

I have created the json at the following URL based on your provided sample.  Please let me know if you see any issues and I will work to address them:
https://dakboard.com/site/download/raspberrypi

XECDesign, 2023-05-03:
I didn't dig too much, but it looks like there's a default username and password combination, the image contains ssh host keys and ssh is on by default. The filesystem also contains windows System Volume Information directories, fsck .rec files, macos .DS_Store files, all the logs containing the commands that were run to configure the image and so on.

cocarrig, 2023-05-03:
Hi @XECDesign,

Thank you for those details, I will take this back and establish a path forward to address your call outs. I'll have an update when its ready.

ghollingworth, 2023-05-10:
We're trying to encourage people to stick to the various statutes, regulations and laws which are either already in force, for example, the PSTI bill in the UK which has now been passed or will become law soon enough.

https://bills.parliament.uk/bills/3069

This specifically calls for no default credentials on a number of IoT devices, Raspberry Pis are specifically used in exactly this scenario and we're trying to help people better understand the reasoning behind these rules...

Basically, if you've got a DAKboard at home plugged into the wall and running, and the user enables ssh there is nothing stopping a hacked IoT device, (a security camera for example) from accessing your device and along with it the credentials (cookies) stored on it...  All you need to do is to remove the defaults (most importantly the password) and set it when they request to enable it.

cocarrig, 2023-05-10:
Hi @ghollingworth,

Thank you for clarifying, I will update our requirements,  Thank you!

cocarrig, 2023-05-19:
@ghollingworth and Team,

The image has been updated at the previously shared link to address the callouts.  We did decide to set a unique password for the user when the image first initializes to better protect the system if the user decides to enable VLC in the GUI.  VLC is added for multiple display support (forces the use of the legacy style desktop needed for some monitors).  SSH is disabled at initialization.  SSH can be enabled by the user, as well as setting a password only after enabling SSL and password protecting the configuration page.  

If there are other updates you would like completed down the road you can reach me at ccarrigan@dakboard.com or the team at contact@dakboard.com as well.

XECDesign, 2023-05-19:
Could you please provide the snippet of json you would like added to [os_list_imagingutility_v3.json](https://downloads.raspberrypi.com/os_list_imagingutility_v3.json)? It should reference the json file you linked to earlier.

cocarrig, 2023-05-19:
Hi @XECDesign,

I believe this covers the required fields:
`{
    "name": "DAKboard",
    "description": "DAKboard is a customizable display for photos, calendar, news, weather and so much more! DAKboard makes it easy to get organized so you won’t miss a thing.",
    "website": "https://dakboard.com",
    "icon": "https://static.dakboard.com/assets/img/dakboard-logos/dakboard-logo-favicon-large.png",
    "subitems_url": "https://dakboard.com/site/download/raspberrypi
    "
},`

XECDesign, 2023-05-25:
As far as I can tell, the image still ships with the problems mentioned earlier. But, a few seconds after the pi is fully booted, the password is changed, ssh keys are removed and ssh is disabled. So there's still a good chunk of time you can ssh in and have full root access, no?

@ghollingworth Does that block it from being added ro rpi-imager?

XECDesign, 2023-05-25:
Added to the other specific-purpose os menu.

cocarrig, 2023-05-25:
Thank you @XECDesign and @ghollingworth,

I have created an internal case on our end to modify the initialization approach to close out that several second window you indicated.

tdewey-rpi, 2023-11-17:
Closing, as you were added prior to the form. I'll be in touch about our new contact requirements if you've not already had a note.

closed 2023-11-17 by tdewey-rpi