raspberrypi/rpi-imager #386: Error downloading OS list from Internet

mdavidjohnson, 2022-03-07:
On Win 10 Home 64-bit the Raspberry Pi Imager worked perfectly with the first SD card. But when I tried to image a second card, I got "Error downloading OS list from Internet" on starting the Imager and it continues to report this with every attempt to restart the Imager; even after re-downloading and re-installing the Imager. I've been trying and re-trying it for over a half-hour now.

maxnet, 2022-03-07:
You do can access the following URL with your normal web browser?
https://downloads.raspberrypi.org/os_list_imagingutility_v3.json

Is there any parental control, ad filtering, proxy server, security software, or other software that can restrict your Internet access active?

mdavidjohnson, 2022-03-07:
No problem at all navigating to that json page. Meanwhile, however, I went to https://www.raspberrypi.com/software/ and downloaded the 64-bit zip file for manual installation. After extracting the img file, the Imager successfully installed the software to the card from the img file's location on my Win 10 hard disk.

mdavidjohnson, 2022-03-07:
I put the full 64-bit Raspberry Pi OS on one SD card, and the 64-bit Lite OS on another. Both worked properly in the Pi 4.

jiapei100, 2023-07-19:
> You do can access the following URL with your normal web browser? https://downloads.raspberrypi.org/os_list_imagingutility_v3.json
> 
> Is there any parental control, ad filtering, proxy server, security software, or other software that can restrict your Internet access active?

where to put this `.json` file in order to let `rpi-imager` find those operating systems?

I know that I can manually download respective **OS**s... But, prefer if `rpi-imager` is able to find those **OS**s for me...

yuukiAme, 2023-10-12:
It looks like https://downloads.raspberrypi.org went down because the Raspberry Pi Imager is reporting Error downloading OS list from Internet.

tdewey-rpi, 2024-02-06:
Closing as stale. No sign there's an ongoing issue here, and no updates since Oct '23. Please try with the latest release before opening a new issue.

closed 2024-02-06 by tdewey-rpi