raspberrypi/rpi-imager #389: [OS Request] UbuntuDockerPi

guysoft, 2022-03-09:
Ready to use 64bit ARM Docker for Raspberry Pi running on Ubuntu

pi-imager PR: https://github.com/guysoft/pi-imager-web/pull/3

This image in many cases used by people to build stuff for arm64.
My telemetry has ~100 people using it now. Might be more I can't see.

```json
    "os_list": [
        {
            "name": "UbuntuDockerPi",
            "description": "Ready to use 64bit ARM Docker for Raspberry Pi running on Ubuntu",
            "icon": "https://raw.githubusercontent.com/guysoft/CustomPiOS/devel/media/rpi-imager-CustomPiOS.png",
            "website": "https://github.com/guysoft/UbuntuDockerPi",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-ubuntudockerpi.json"
        }
    ]
```

lurch, 2022-03-10:
"Ready to use 64bit ARM Docker for Raspberry Pi running on Ubuntu" makes it sound like an image that you'd run on your desktop Ubuntu PC, rather than on your Raspberry Pi? :confused: (i.e. perhaps it needs rewording slightly)

guysoft, 2022-03-10:
@lurch
Sure enough, fixed

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi