raspberrypi/rpi-imager #390: Setting up ssh password login does not work

dannenberg, 2022-03-09:
I'm using Imager v1.7.1 and the 64-bit Raspian OS from the Raspberry Pi OS (Other) section. I check Enable SSH and select Use password authentication. I check Set username and password, leave the default Username of pi and set a simple password. The install is reachable via hostname, but attempts to log in via ssh pi@HOSTNAME.local fail when I input the password.

Not sure if this is an imager bug or a 64bit image bug, but I'd love to see it fixed.

maxnet, 2022-03-09:
Unable to reproduce.

```
$ ssh pi@pi64.local
The authenticity of host 'pi64.local (192.168.178.77)' can't be established.
ECDSA key fingerprint is SHA256:E7UE8c9CuF3yXuxeczLxxXul8n86cCuHnyul8h8M/Ps.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'pi64.local' (ECDSA) to the list of known hosts.
Warning: the ECDSA host key for 'pi64.local' differs from the key for the IP address '192.168.178.77'
Offending key for IP in /home/max/.ssh/known_hosts:289
Are you sure you want to continue connecting (yes/no)? yes
pi@pi64.local's password: 
Linux pi64 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@pi64:~ $ 
```

dannenberg, 2022-03-09:
Hmm... A few colleagues have encountered the same problem. We are using the Mac OS X imager in case that is a factor and the full (not-lite) version of the 64-bit OS. We are also using the wireless connection in case that is a factor. I'll try to think of other possible variables here and get back to you.

maxnet, 2022-03-09:
Works fine on a Mac as well:

[image]

You do are absolutely positive you specified an unique .local hostname, and aren't accidentally ssh'ing to your co-workers Pi, right?

dannenberg, 2022-03-09:
Yes, wholly unique hostnames. I'll try this again from another machine in a moment. Thanks for the quick responses. I too did not believe this was possible when it was first brought to me.

dannenberg, 2022-03-09:
Terribly sorry. Turns out they'd been changing the username and then ssh'ing in as pi. Thanks for your time and attention. I really appreciate your rapid responses.

closed 2022-03-09 by dannenberg