raspberrypi/rpi-imager #391: Reaches out to stats server without informing user - GDPR violation?

flatsiedatsie, 2022-03-10:
Just a quick heads-up. I noticed that the latest version of the Raspberry Pi imager was trying to reach out to a statistics server after I hit the 'write' button.

This means you are receiving IP addresses, which is generally regarded as personal information. Only actually storing 'anonymous usage data' does not change this? Fact is: I can only guess what you're collecting.

I did not notice any privacy policy I had to accept before downloading or using the software, so... that might be a little dubious? Or at the very least a little rude?

Granted, downloading disk images by itself implies that a server connection is made...

ghollingworth, 2022-03-10:
We do not capture or save any IP addresses, so GDPR doesn't apply.  You can see our description of the telemetry in the Readme...  You can also read the source to see what we're saving (a number of counters and that's it)

closed 2022-03-10 by ghollingworth

flatsiedatsie, 2022-03-11:
It would still be proper to inform people of this and give them a choice to opt-out of sharing telemetry before they use the software.