raspberrypi/rpi-imager #392: [OS Request] MtigOS

guysoft, 2022-03-10:
A Rpi distro that lets you receive, store and graph sensor information from ESP8266 chips or anything that pushes to MQTT.

It uses and MTIG stack: [Mosquitto](https://mosquitto.org/), [Telegraf](https://www.influxdata.com/time-series-platform/telegraf/), [InfluxDB](https://www.influxdata.com/) and [Grafana](https://grafana.com/) which are all pre-configured to work together. They automatically update using Docker.

It has less users than the last distros I put for request, however I think it is something that you would like to consider posting here. 
Its a common usage pattern for Pis to receive sensor information from Mqtt. 
Here is some [coverage about it](http://www.whatimade.today/display-colourful-live-graphs-on-any-raspberry-pi-with-just-a-few-clicks-2/). 
Here is [my own post about it](https://guysoft.wordpress.com/2020/03/02/mtigos/).

```json
    "os_list": [
        {
            "name": "MtigOS",
            "description": "Raspberry Pi distro that lets you receive, store and graph sensor information from ESP8266 chips",
            "icon": "https://raw.githubusercontent.com/guysoft/CustomPiOS/devel/media/rpi-imager-CustomPiOS.png",
            "website": "https://github.com/guysoft/MtigOS",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-mtigos.json"
        }
    ]
```

NonstickAtom785, 2022-03-11:
Hm. @guysoft I had a large list of OS Requests to add here as well. Do you think we should maybe just make an OS request list? Under one name and have an optional formatting?

guysoft, 2022-03-11:
I don't manage the repo. I do think that each distribution should be evaluated separately.

What I did in my rpi-imager fork is have a repo that you PR the json, and then it builds the list automatically:

https://github.com/guysoft/pi-imager-web

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi