raspberrypi/rpi-imager #394: Advanced Settings script seem to not be working

lertsoft, 2022-03-11:
The script that advance settings create for the raspberry pi seems to not be working.
I flash three different RPI OS-es: recommended 32 bits, 32 bits without GUI and 64 bits and it always fail to let me access the RPI through SSH after waiting a minimum of 5 minutes.

I don't necessarily know whats the issue as the script is created and has from what I can tell all the parameters that i configure on advance settings  but it seems that the RPI does not execute it.

Using latest version of RPI-IMAGER on a 2021 macbook pro with m1, and MAC OS Monterey (12.2.1)

maxnet, 2022-03-11:
>it always fail to let me access the RPI through SSH after waiting a minimum of 5 minutes.

With what error?

>I don't necessarily know whats the issue as the script is created and has from what I can tell all the parameters that i configure 
>on advance settings but it seems that the RPI does not execute it.

And what are all the parameters you set?

You are not able to hook up a HDMI display to check if at least some of those parameters are set?
Just that you are unable to connect over wifi (if you are using that) to SSH, does not necessarily mean that no settings are being set.

lertsoft, 2022-03-14:
> With what error?
I did not see any errors on rpi-imager, and could not see any errors on the Raspberry pi because i am doing a headless setup.

> And what are all the parameters you set?
I set:

maxnet, 2022-03-14:
>I did not see any errors on rpi-imager

Meant with what error does your ssh client fail to connect.

>i am doing a headless setup.

That makes such things hard to debug...

The Pi does "consume" firstrun.sh?
If you reinsert the SD card in the reader of your Mac after writing, there is a firstrun.sh, and if you put it in the Pi, let it boot for a while, and then put it back in the SD card reader of your Mac the firstrun.sh is gone?

lertsoft, 2022-03-16:
nope, the pi did not consume the firstrun.sh

maxnet, 2022-03-16:
>nope, the pi did not consume the firstrun.sh

If you open `cmdline.txt` in a text editor, Imager did successfully add " systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target" to the same line as the other options there?

lertsoft, 2022-03-17:
it did not

maxnet, 2022-03-17:
>it did not

If you uncheck "Eject media when finished", and leave the card in the reader after the write.
Does cmdline.txt change successfully then?

lertsoft, 2022-03-18:
The script seems to be working right now. The only thing that I did was used the shortcut cmd+shift+x to open the settings. ssh is working, WLAN is working, I haven't checked on the hostname and any of the other advanced settings.

lertsoft, 2022-03-18:
Also i did not update or reinstall the program. But the program did refresh and I force quit it before everything ran.

trasba, 2022-05-05:
> > nope, the pi did not consume the firstrun.sh
> 
> If you open `cmdline.txt` in a text editor, Imager did successfully add " systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target" to the same line as the other options there?

For me this helped me figure out the issue. I cannot confirm if not ejecting the drive automatically in the end would solve the problem. But I could solve it by hand.
I am running the imager on a windows 11 OS.
Maybe this is worth a note in the documentation.

PaoloRondot, 2022-07-23:
I'm running in the exact same issue as lertsoft...
Does not consume firstrun.sh. I tried multiple time to flash with 64 bits lite version and it doesn't work

tdewey-rpi, 2024-02-06:
I'd encourage the OP to retest with 1.8.5. We had a known issue with the first character of the SSH password getting dropped which has since been resolved.

tdewey-rpi, 2024-08-28:
Closing as 'Stale'.

closed 2024-08-28 by tdewey-rpi