raspberrypi/rpi-imager #396: Can't flash any images

FlyinPancake, 2022-03-13:
OS: Fedora 35
Version: 1.7.1

To reproduce: 
1. Install from flatpak (fedora or flathub) or rpm (via dnf)
2. Run as unprivileged user
3. Select os and SD card device, set settings etc. 
4. Click on flash, enter password in the polkit pop-up.
5. Still no access to the device :( 
[image]

When I run rpi-imager as root everything works as expected.

maxnet, 2022-03-13:
>OS: Fedora 35

Running on a Pi or some laptop?
(External USB based readers usually show up as /dev/sdX instead of /dev/mmcblk0)

Any security constraints such as SELINUX active?

If you keep "sudo busctl monitor --system" running, and then attempt to use Imager.
At the time you get the "cannot open storage device" message, does it show any detailed error message coming from udisks2 there?
(It may output a lot, but the error messages should be highlighted red in the output, so should stand out.)

FlyinPancake, 2022-03-14:
Running on a laptop with selinux enabled. 

I'm going to post the logs in a short time.

Frivolous2, 2022-03-29:
I have the same issue. Fedora 35 Silverblue, using latest Imager from Flathub.

Here is the output of sudo busctl monitor --system: [logs.txt](https://github.com/raspberrypi/rpi-imager/files/8375603/logs.txt)

jeremiah-miller, 2022-09-24:
Same issue on arch + sway. Fortunately what @kees-closed suggested worked

maxnet, 2022-09-27:
>Same issue on arch + sway. 

Sway has a different issue of not having a polkit agent to ask the password. 
That other issue is described here: https://github.com/raspberrypi/rpi-imager/issues/359

The issue on this page is different in that that it does prompt for password (so there must be a polkit agent), but the udisks2 service disconnects from DBus (probably crashes) due to unknown reasons.

avinal, 2022-12-29:
Facing issue on Fedora 37 
[image]

sheerun, 2023-08-14:
For some reason mmcblk device (sd card reader) doesn't work for me, while USB adapter works...

maxnet, 2023-08-27:
>It would be great if the Raspberry Pi devs would fix this

Well, it doesn't affect the Linux distributions we support, being RPI OS and Ubuntu.
Aren't you associated with the Fedora project affected? ;-)

maxnet, 2023-08-27:
>The README doesn't state any special configuration or requirement for udisks2 to work properly.

It is not a hard requirement.
As you can also grant users access to storage in other ways.

E.g. if you are not too concerned about security just adding the current user to disk group will probably also "solve" it...

```
sudo usermod -a -G disk $USER
```

(logout and in again afterwards)

Hallo32, 2023-11-19:
@maxnet 

A way how to reproduce the problem with Pi OS and raspberry Pi.

* Get a raspberry Pi  5, a micro SD card and an USB Stick
*  Write Pi OS to the USB stick
* Start the Raspberry Pi 5 from the usb stick
* Install rpi-imager as Debian package
* Try to write an Pi OS image  to the micro SD card inside the raspberry Pi 5m which was booted from the USB stick.
* The error message appears.

maxnet, 2023-11-19:
>ssh -X

Starting rpi-imager over X forwarding is not a supported scenario.
Cannot ask the OS (polkit) to prompt user for credentials without working dbus and desktop session.

Hallo32, 2023-11-19:
> Starting rpi-imager over X forwarding is not a supported scenario. Cannot ask the OS (polkit) to prompt user for credentials without working dbus and desktop session.

Can the missing polkit be detected and an error message created, which is showing the correct error source?

Thanks for the fast reply time :)

maxnet, 2023-11-19:
Problem is that users than also expect solutions. And the workarounds necessary are very Linux distribution/desktop environment specific.

E.g. we may suggest that "sudo -g disk rpi-imager" does work.
But it may not, if the storage you want to write to was mounted by someone other than the current user.

Hallo32, 2023-11-19:
I would go a step back in such a situation and inform the user, that the software has been used in a not expected way and list the official supported ways.
As you mention, you can not handle all cases on all the distributions and update this information, if the process has changed.

In the end I have used the --cli option.

JonHeron, 2023-12-18:
Neither of these solutions have worked for me on F38
Its like it resets the connection when you execute the write
The built in image writer fails in the same way 
Trying dd now, very slow...

JonHeron, 2023-12-18:
It worked after disabling selinux

closed 2024-02-06 by tdewey-rpi

Jarogniew13k, 2025-03-10:
the problem is selinux. If you turn off selinux with comand "sudo setenforce 0"  it will work

tdewey-rpi, 2025-03-17:
Unfortunately, I have to agree with @kees-closed.

**Do not disable selinux** if your distro uses it out of the box, for _any_ application. Instead, report the to distributor for that distro. In order to avoid this becoming a zombie issue, I'm going to now restrict it to contributors.