raspberrypi/rpi-imager #397: Update macOS Icon to Big Sur Style

Logicer16, 2022-03-16:
Since the release of macOS Big Sur back in 2020, Apple has redesigned many of their app's icons to use a rounded square design, and many third-party apps have followed suit.

The Raspberry Pi Imager currently looks out of place amongst other apps that have already adopted this style. 

Apple has provided resources for this new style in their [Human Interface Guidelines](https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/app-icon/)

elsiehupp, 2023-06-09:
I have made Big-Sur-style icons for other projects, and most recently I merged this template over at the Nextcloud Desktop repository:

[image]

[Nextcloud-macOS-icon.svg](https://github.com/nextcloud/desktop/blob/04d16b7824e633b66fbd9fbbed98c615e14e8c2d/theme/colored/Nextcloud-macOS-icon.svg?short_path=44dd0e3)

This is a 100% vector Big-Sur-style icon, layer effects and all, and the above link should go to the code view, not the image preview.

(The annotation formatting apparently [broke the build process](https://github.com/nextcloud/desktop/pull/5757) in way that wasn't caught by the continuous integration, though...)

Anyway as you can see in the code annotation, I (as the author of the SVG document) released the template under a CC0 Public Domain Dedication, so hopefully this can be some help in creating a Big-Sur-style version of the RPI Imager icon!

(I should also note that the template is versatile enough that it should be useful for icon shapes other than the Apple "Squircle"—the annotations explain which SVG element to replace for a different shape.)

Logicer16, 2023-11-30:
Closed in #695

closed 2023-11-30 by Logicer16