raspberrypi/rpi-imager #399: [OS request] Lakka

ToKe79, 2022-03-18:
Hello good people!

We, the maintainers of Lakka, would be very pleased if you can include Lakka - JEOS for RetroArch in the Raspberry Pi Imager OS selector in the `Emulation and game OS` section. Here is the snippet to add our `.json` with available latest images:

```
 "os_list": [
        {
            "name": "Lakka",
            "description": "The DIY retro emulation console",
            "icon": "https://nightly.builds.lakka.tv/.rpi-imager/icon.png",
            "website": "https://lakka.tv/",
            "subitems_url": "https://nightly.builds.lakka.tv/.rpi-imager/rpi-imager.json"
        }
    ]
```
I tested the `subitems_url` with `rpi-imager.exe --repo https://nightly.builds.lakka.tv/.rpi-imager/rpi-imager.json` on my Windows box:

[image]

Sources to the OS: https://github.com/libretro/Lakka-LibreELEC
Website of the OS: https://lakka.tv/

Thank you for your consideration.

XECDesign, 2022-03-18:
Added

closed 2022-03-18 by ToKe79

mudge, 2023-04-18:
Hi @ToKe79, would it be possible for you to update https://nightly.builds.lakka.tv/.rpi-imager/rpi-imager-devbuild.json to include valid JSON even when it is empty?

It's currently returning an empty response with a 200 OK status code but can't be parsed as JSON. To match the other operating system lists, perhaps you could return the following if there are no images available?

```json
{"os_list": []}
```

ToKe79, 2023-04-18:
Sorry for that @mudge, we ran out of space (hence zero sized json). New json file is being generated as I write this. I will also make sure that in case there are write errors while creating new json file, it will not replace the current one.

deneraraujo, 2025-03-12:
Missing on current version (1.8.5)

ToKe79, 2025-03-13:
I posted a request to add the OS to the list a couple weeks ago, but there is still no feedback from the maintainers.